Koenkk/zigbee2mqtt #8218: Temperature & humidity sensor reports only when button was pressed

adriannowaq, 2021-08-02:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/8217

<div type='discussions-op-text'>

<sup>Originally posted by **adriannowaq** August  2, 2021</sup>
I have a trouble with Temperature & Humidity Sensor from Tuya,:
Model on the box: SNTZ002
Auto-recognized model in Zigbee2Mqtt:  RH3052

What's wrong: I receive data only in two cases: 1. When I restart coordinator, 2. When I push the button on device (Device is spamming for 3 seconds and then it goes to sleep.

Tried to make something like this: 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

 const definition = {
    zigbeeModel: ['RH3052'],
    model: 'RH3052',
    vendor: 'TUYATEC-yg5dcbfu',
    description: 'Temperature & humidity sensor with display',
    supports: 'temperature and humidity',
    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement'];
        await bind(endpoint, coordinatorEndpoint, binds);
        await configureReporting.temperature(endpoint, {min: 1, max: 5, change 1});
    },
};

module.exports = definition;
```

But when I restart Zigbee2Mqtt I'm getting an error:

> Aug 02 09:33:54 raspberrypi npm[14378]: Zigbee2MQTT:info  2021-08-02 09:33:54: Disconnecting from MQTT server
> Aug 02 09:33:54 raspberrypi npm[14378]: Zigbee2MQTT:info  2021-08-02 09:33:54: Stopping zigbee-herdsman...
> Aug 02 09:33:54 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=killed, status=15/TERM
> Aug 02 09:33:54 raspberrypi npm[14378]: Zigbee2MQTT:info  2021-08-02 09:33:54: Stopped zigbee-herdsman
> Aug 02 09:33:54 raspberrypi npm[14378]: Zigbee2MQTT:info  2021-08-02 09:33:54: Stopped Zigbee2MQTT
> Aug 02 09:33:54 raspberrypi systemd[1]: zigbee2mqtt.service: Succeeded.
> Aug 02 09:33:54 raspberrypi systemd[1]: Stopped zigbee2mqtt.
> Aug 02 09:33:54 raspberrypi systemd[1]: Started zigbee2mqtt.
> Aug 02 09:33:55 raspberrypi npm[14469]: > zigbee2mqtt@1.20.0 start /opt/zigbee2mqtt
> Aug 02 09:33:55 raspberrypi npm[14469]: > node index.js
> Aug 02 09:33:58 raspberrypi npm[14469]: (node:14481) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected number
> Aug 02 09:33:58 raspberrypi npm[14469]: (Use `node --trace-warnings ...` to show where the warning was created)
> Aug 02 09:33:58 raspberrypi npm[14469]: (node:14481) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
> Aug 02 09:33:58 raspberrypi npm[14469]: (node:14481) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
> 
> 

Version of Zigbee2Mqtt: 1.20.0
Coordinator version: 20190608
</div>

nurikk, 2021-08-03:
you need to add a colon ":"  after `change`
```
        await configureReporting.temperature(endpoint, {min: 1, max: 5, change: 1});
```

adriannowaq, 2021-08-03:
> you need to add a colon ":" after `change`
> 
> ```
>         await configureReporting.temperature(endpoint, {min: 1, max: 5, change: 1});
> ```

added colon, but it does not start:

> Aug 03 10:14:36 raspberrypi systemd[1]: Started zigbee2mqtt.
> Aug 03 10:14:37 raspberrypi npm[3593]: > zigbee2mqtt@1.20.0 start /opt/zigbee2mqtt
> Aug 03 10:14:37 raspberrypi npm[3593]: > node index.js
> Aug 03 10:14:39 raspberrypi npm[3593]: (node:3607) UnhandledPromiseRejectionWarning: AssertionError [ERR_ASSERTION]: Converter field exposes is undefined
> Aug 03 10:14:39 raspberrypi npm[3593]:     at validateDefinition (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:55:16)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at Object.addDefinition [as addDeviceDefinition] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:86:5)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.js:13:38)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at new Controller (/opt/zigbee2mqtt/lib/controller.js:77:34)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at start (/opt/zigbee2mqtt/index.js:42:18)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:60:1)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at Module._compile (internal/modules/cjs/loader.js:1085:14)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at Module.load (internal/modules/cjs/loader.js:950:32)
> Aug 03 10:14:39 raspberrypi npm[3593]:     at Function.Module._load (internal/modules/cjs/loader.js:790:14)
> Aug 03 10:14:39 raspberrypi npm[3593]: (Use `node --trace-warnings ...` to show where the warning was created)
> Aug 03 10:14:39 raspberrypi npm[3593]: (node:3607) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
> Aug 03 10:14:39 raspberrypi npm[3593]: (node:3607) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
>

adriannowaq, 2021-08-03:
> you need to add a colon ":" after `change`
> 
> ```
>         await configureReporting.temperature(endpoint, {min: 1, max: 5, change: 1});
> ```

Fixed that, i also added expose, now it looks like this:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

 const definition = {
    zigbeeModel: ['RH3052'],
    model: 'RH3052',
    vendor: 'TUYATEC-yg5dcbfu',
    description: 'Temperature & humidity sensor with display',
    supports: 'temperature and humidity',
    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.humidity(), e.temperature(), e.battery()],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement'];
        await bind(endpoint, coordinatorEndpoint, binds);
        await configureReporting.temperature(endpoint, {min: 1, max: 5, change: 1});
    },
};

module.exports = definition;
```
but still I'm getting and error:

> Aug 03 10:28:10 raspberrypi npm[5741]: Zigbee2MQTT:error 2021-08-03 10:28:10: Failed to configure 'kitchen', attempt 2 (ReferenceError: bind is not defined
> Aug 03 10:28:10 raspberrypi npm[5741]:     at Object.configure (/opt/zigbee2mqtt/lib/util/externally-loaded.js:22:9)
> Aug 03 10:28:10 raspberrypi npm[5741]:     at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:137:45)
> Aug 03 10:28:10 raspberrypi npm[5741]:     at Configure.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/configure.js:119:18)
> Aug 03 10:28:10 raspberrypi npm[5741]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))
>

nurikk, 2021-08-03:
your code is missing this part
```
const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

```

add it before ` const definition = {` piece

adriannowaq, 2021-08-04:
Thanks, now its ok:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

 const definition = {
    zigbeeModel: ['RH3052'],
    model: 'RH3052',
    vendor: 'TUYATEC-yg5dcbfu',
    description: 'Temperature & humidity sensor with display',
    supports: 'temperature and humidity',
    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.humidity(), e.temperature(), e.battery()],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement'];
        await bind(endpoint, coordinatorEndpoint, binds);
        await configureReporting.temperature(endpoint, {min: 1, max: 5, change: 1});
    },
};

module.exports = definition;
```

What about repeated messages? It looks like this:

```
08/04/2021 8:07:17 AM 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}
08/04/2021 8:07:17 AM(-0.01 seconds) 
{"battery":81,"humidity":48.12,"linkquality":134,"temperature":24.05,"voltage":2800}

08/04/2021 8:02:20 AM(-4.96 minutes) 
{"battery":81,"humidity":48.12,"temperature":24.05,"voltage":2800}

08/04/2021 8:01:18 AM(-61.8 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0.05 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
08/04/2021 8:01:18 AM(-0 seconds) 
{"battery":81,"humidity":48.12,"linkquality":136,"temperature":24.05,"voltage":2800}
```