Koenkk/zigbee2mqtt #8222: problem starting after upgrade to 1.21.0

mabi, 2021-08-02:
## What happened

After update to 1.21.0 zigbee2mqtt does not start, initial build fails with the message

> Building Zigbee2MQTT... (initial build), failed

## What did you expect to happen

I expected a correct start

## How to reproduce it (minimal and precise)

npm start

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: CC2531
Adapter firmware version: 20190315

Node version v10.15.3

Hardware: raspberry pi 3

> Building Zigbee2MQTT... (initial build), failed
(node:9611) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! zigbee2mqtt@1.21.0 build: `tsc && node index.js writehash`
npm ERR! Exit status 2
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.21.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

> npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-08-02T13_22_01_516Z-debug.log

>    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
(node:9611) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:9611) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Koenkk, 2021-08-02:
Can you run `npm run build` in the Zigbee2mqtt directory?

mabi, 2021-08-02:
This way it worked, `build `returned no error and `start` was successful, thank you.

Koenkk, 2021-08-02:
I think I found the issue. Can you switch to the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), remove `dist/.hash` and start z2m (does it start correctly now)?

mabi, 2021-08-02:
> I think I found the issue. Can you switch to the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), remove `dist/.hash` and start z2m (does it start correctly now)?

yes, it starts correctly, thanks.

closed 2021-08-02 by Koenkk

nefh, 2021-08-02:
Thanks for quick update, on FreeBSD 12 I also had problem with `dist/.hash`, dev branch solved this issue. The message was a bit different: 

> Building Zigbee2MQTT... (initial build), finished
(node:61056) UnhandledPromiseRejectionWarning: Error: ENOENT: no such file or directory, open 'dist/.hash'
    at Object.openSync (fs.js:498:3)
    at Object.readFileSync (fs.js:394:35)
    at checkDist (/opt/zigbee2mqtt/index.js:61:25)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async start (/opt/zigbee2mqtt/index.js:69:5)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:61056) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:61056) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

One note more - since this update starting via rc.d script needed /usr/local/sbin:/usr/local/bin in $PATH, otherwise it could not find npm. I made a quick & dirty fix by putting PATH directly into `/usr/local/etc/rc.d/zigbee2mqtt` but I'm sure there is better way for this :)

```

#!/bin/sh

# PROVIDE: zigbee2mqtt
# REQUIRE: LOGIN
# KEYWORD: shutdown

# zigbee2mqtt_enable="yes"

. /etc/rc.subr

PATH=/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin

name=zigbee2mqtt
rcvar=zigbee2mqtt_enable

zigbee2mqtt_pid=/var/run/zigbee2mqtt.pid
zigbee2mqtt_log=/var/log/zigbee2mqtt_daemon.log

# Can also be -9 etc. -HUP usually will only cause sadness.
sig_stop=-KILL

start_cmd="zigbee2mqtt_start"
stop_cmd="zigbee2mqtt_stop"
restart_cmd="zigbee2mqtt_restart"    # didn't bother to implement

load_rc_config ${name}

# Set to either 'node' to use default or set EXPLICIT e.g. '/usr/local/bin/node1.10'
command="/usr/local/bin/node"
# Command path must always be explicit.
command_script="/opt/zigbee2mqtt/index.js"
# Contain arguments in quotes, use ' and \"
command_args=""

zigbee2mqtt_start()
{
echo $PATH
    # See daemon(8) for more details.
    daemon -f -p $zigbee2mqtt_pid -o $zigbee2mqtt_log $command $command_script $command_args
    if [ $? -ne 0 ]; then
        echo "Error starting zigbee2mqtt."
        exit 1
    fi
    echo "Starting zigbee2mqtt."

}

zigbee2mqtt_stop()
{
    echo "Stopping zigbee2mqtt."
    kill $sig_stop `cat $zigbee2mqtt_pid`
}

zigbee2mqtt_restart()
{
    zigbee2mqtt_stop;
    zigbee2mqtt_start;
}

run_rc_command "$1"
```