Koenkk/zigbee2mqtt #8226: TS0203 unsupported

ggdijkstra, 2021-08-02:
The last update and TS0203 unsupported !
[image]

jimmyeao, 2021-08-02:
Same here, all the TS0203 devices now show as unsupported

Koenkk, 2021-08-02:
Added `_TYZB01_ncdapbwy` back.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-02 by Koenkk

jimmyeao, 2021-08-03:
Updated the dev branch this morning, it still shows as unsupported

robert2s, 2021-08-08:
Is a duplicate of #8266 and fixed in dev version (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). I had the same problem, and all TS0203 reappeared instantly today (2021-8-8) after I switched to add-on zigbee2mqtt **Edge** in Home Assistant. This is what I did:

1. Disable add-on zigbee2mqtt: In the Information page of zigbee2mqtt in Home Assistant, stop the add-on, and put the "start on boot" (Dutch: start bij opstarten) to Off.
2. Install add-on zigbee2mqtt **Edge**: in the add-on Store of Home Assistant, select zigbee2mqtt **Edge**, and install it.
3. Copy the configuration from zigbee2mqtt to zigbee2mqtt **Edge**:  in the Configuration page of zigbee2mqtt, copy the complete yaml text (ctrl-C). Go to the Configuration page of zigbee2mqtt **Edge**, and paste it there.
4. Start add-on zigbee2mqtt **Edge**, and switch on "start on boot" and "show in sidebar".

As soon as a new version of zigbee2mqtt is available (i.e. version 1.22.0) I recommend to reverse the above procedure, so you use the normal zigbee2mqtt add-on again.
Note: alternatively, you may decide to wait until version 1.22.0 is availble, because that will automatically solve the problems.