Koenkk/zigbee2mqtt #8227: Very high latency with an outdoor sonoff zbmini

ahocquet, 2021-08-02:
# Bug Report
## What happened

Hello,

I own a ZZH coordinator adapter backed up with five CC2530 DC powered routers (homemade built thanks to the guide on z2m website).

I have placed on of these router outside close to my garden sheds, 10m away from my kitchen. In this garden shed, I have a Sonoff ZBMINI that I use to control some lights.

The problem is that I have a very high latency (more than 10s) controlling this zonoff zbmini switch. I have other zbmini switches indoor without any problems.

I've mapped my setup and here's the result. My outdoor zbmini name is "Eclairage guirlande" whereas my CC25430 router close to it is named "CC2530 Router E - Local piscine"

[image]
[image]
[image]

The funny thing is that when I restart Z2M, then the first time I activate the switch it reacts quite fast. But then it starts to lags a lot, like if it was trying to use a slower route than the one with the closest router "CC2530 Router E - Local piscine". I've red-circled the other possible routers that might be used for this switch but that are way further in distance.

[Logs here](https://hastebin.com/ufujoqayir.apache)
## What did you expect to happen
I'd expect a low latency around 1 second.

## How to reproduce it (minimal and precise)
Just using it.

## Debug Info
Zigbee2MQTT version: 1.20.0-dev commit: 46e2a77
Adapter hardware: ZZH coordinator and five CC2530, routers
Adapter firmware version: 
 - CC2652R_coordinator_20210120
 - CC2530_CC2591_router_2020_09_29

emms007, 2021-08-23:
I am having the same issue, using two ZBMINI that are pretty close in an garden electrical box (10cm away).
Some action would go there pretty fast, some other would timeout.

I wonder if the problem is not that the routers are not too close, as I am reading in the solution of https://github.com/Koenkk/zigbee2mqtt/issues/3748

How close is the distance of routers in your case?

closed 2021-09-30 by github-actions[bot]

ahocquet, 2021-12-09:
@emms007 sorry for the late reply. I didn't want to spend to much time in solving this issue so I've replaced these sonoff zbmini with shelly relays. Problem solved :)