Koenkk/zigbee2mqtt #8228: Map link problem with the coordinator

larod241, 2021-08-03:
## What happened

I have just redo the installation of all my sensors taking advantage of making the FW maj of the Combee II and Zigbee2MQTT (1.21.0)
After a few hours, all the coordinator-related sensors are no longer related in the MAP display.
They still work however

They were all a few hours before!

I restarted Ha Core but nothing ...

Just after the pairing
[image]

After a while 
[image]

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: Conbee II
Adapter firmware version: 0x266f0700

nukleuz80, 2021-08-06:
Hi,

i have the same setup and same issue you subscribed.

Koenkk, 2021-08-06:
I've added an explanation on why this happens to the FAQ: https://www.zigbee2mqtt.io/information/FAQ.html#why-are-some-links-missing-from-my-networkmap . This is not something that can be fixed from zigbee2mqtt.

closed 2021-08-06 by Koenkk

xekil, 2021-08-18:
same problem, and thanks for the FAQ

In my case, I have xiaomi plugs which are not Zigbee 2 and which nevertheless behave like this, the same for certain xiaomi modules which are directly with the coordinator.