Koenkk/zigbee2mqtt #8231: unsupported device with Zigbee Tuya Smart Door Window Sensor Magnetic Graffiti after update

tiimsvk, 2021-08-03:
After reinstall hassio and update this device 
https://www.aliexpress.com/item/1005002018443929.html?spm=a2g0s.9042311.0.0.27424c4d4U5yPz
not supported
What has happened?

ggdijkstra, 2021-08-03:
Friendly name
Sensor Voordeur
Last seen
N/A
Device type
EndDevice
Zigbee Model
TS0203
Zigbee Manufacturer
_TZ3000_ebar6ljy
Support status
Not supported

jimmyeao, 2021-08-03:
Similar issue, updated Zigbee2mqtt dev and my tuya contact sensors all stopped being detected, 
Manufacturer is shown as _TYZB01_ncdapbwy

Koenkk, 2021-08-03:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-03 by Koenkk