Koenkk/zigbee2mqtt #8233: TS0203 still shows unsupported on dev branch

jimmyeao, 2021-08-03:
## What happened
Tuya TS0203 devices showed as unsupported after update.

## What did you expect to happen
They would continue to be supported

## How to reproduce it (minimal and precise)Attempt o add a tS0203 contact sensor, pairing is successful but is not configured.

## Debug info
Zigbee2MQTT version: 
1.21.0-dev commit: 056e07a4
Adapter hardware:  CC2652
Adapter firmware version: zstack 3.0 (20201026)

robertalexa, 2021-08-03:
Is this a duplicate of this?
https://github.com/Koenkk/zigbee2mqtt/issues/8231

If so, it would probably be best if you close this issue and post your problem as part of that thread

jimmyeao, 2021-08-03:
mine is a different manufacturer name - _TYZB01_ncdapbwy, Used to work fine a couple of weeks ago, not sure what has changed to break it

Koenkk, 2021-08-03:
Did you do a `npm ci` after updating to the latest dev? `_TYZB01_ncdapbwy` is supported: https://github.com/Koenkk/zigbee-herdsman-converters/blob/2b778ff8443071ecb0b1b1f60ea8c222cc84621e/devices/tuya.js#L17

jimmyeao, 2021-08-04:
Just did another pull and its working again now, thanks

closed 2021-08-04 by jimmyeao