Koenkk/zigbee2mqtt #8234: Help network is not responding anymore

elgerg, 2021-08-03:
I have a network of 33 devices (a mix of wired/router and end points). Last night the network stopped responding so i rebooted my Pi but that didnt help. Some of the devices started working once i had re-flashed the dongle but other devices would not. I cant seem to add new devices to the network either. 

I'm now getting a whole host of errors like:
Error Publish 'set' 'state' to 'LandingFromDownstairs' failed: 'Error: Command 0x5c0272fffedebbf1/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error Publish 'set' 'state' to 'LandingLight' failed: 'Error: Command 0x5c0272fffedebc05/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error Publish 'set' 'state' to 'LoungeCurtains' failed: 'Error: Command 0x5c0272fffed76c38/1 manuSpecificTuya.setData({"status":0,"transid":0,"dp":1,"datatype":4,"length_hi":0,"length_lo":1,"data":[2]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error Publish 'set' 'state' to 'LandingLight' failed: 'Error: Command 0x5c0272fffedebc05/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error Publish 'set' 'state' to 'SpareRoomLight' failed: 'Error: Command 0x5c0272fffec619d5/1 manuSpecificTuya.setData({"status":0,"transid":1,"dp":1,"datatype":1,"length_hi":0,"length_lo":1,"data":[1]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error Publish 'set' 'state' to 'LoungeMainDimmer' failed: 'Error: Command 0x5c0272fffec1fcc7/1 manuSpecificTuya.setData({"status":0,"transid":1,"dp":1,"datatype":1,"length_hi":0,"length_lo":1,"data":[0]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined at Function.fromObject (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:206:17) at Function.fromMireds (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:219:24) at Object.syncColorState (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:744:32) at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:564:51) at /opt/zigbee2mqtt/lib/extension/receive.js:159:41 at Array.forEach (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:158:20) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:408:81) at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:280:24))
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined at Function.fromObject (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:206:17) at Function.fromMireds (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:219:24) at Object.syncColorState (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:744:32) at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:564:51) at /opt/zigbee2mqtt/lib/extension/receive.js:159:41 at Array.forEach (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:158:20) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:408:81) at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:280:24))
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined at Function.fromObject (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:206:17) at Function.fromMireds (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:219:24) at Object.syncColorState (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/color.js:744:32) at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:564:51) at /opt/zigbee2mqtt/lib/extension/receive.js:159:41 at Array.forEach (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:158:20) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:408:81) at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:280:24))

## What happened
Last night the network went offline. I got some weird errors trying to start Zigbee2Mqtt. Things like file/directory doesnt exist (even though the test was successful)

## What did you expect to happen
The network continues to work

## How to reproduce it (minimal and precise)
I'm not sure what failed so i cant figure out whats wrong.

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: CC2531
Adapter firmware version:
Source and default 1.2 firmware tried

Borcon, 2021-08-03:
I get this error message since today. Looks similar.
Zigbee2MQTT version: 1.21.0-1
Adapter hardware: Texas Instruments LAUNCHXL-CC1352P-2
Adapter firmware: 20210708

Debug Publishing 'set' 'state' to 'plug_wlan_dg'
Error Publish 'set' 'state' to 'plug_wlan_dg' failed: 'Error: Command 0x60a423fffe621163/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37006 - 1 - 159 - 6 - 11 after 10000ms)'
Debug Error: Command 0x60a423fffe621163/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37006 - 1 - 159 - 6 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'plug_wlan_dg' failed: 'Error: Command 0x60a423fffe621163/1 genOnOff.on({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 37006 - 1 - 159 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"plug_wlan_dg"},"type":"zigbee_publish_error"}'

robertalexa, 2021-08-03:
Have a look at the Release notes of the latest z2m.
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0

[image]

@Koenkk would probably be the best person to assist you here

Borcon, 2021-08-03:
I read the Release notes. So i stopped and started Z2M serveral times before i flashed to the latest firmware.
But after the firmware update, the network is death.

netweaver1970, 2021-08-03:
I read somewhere, in the tens of daily Z2M issue emails that are floating by in my mailbox, that the backup.json file was only effectively written when the reboot was initiated from the Z2M GUI, not from the manual stop/start service. maybe you ran into that issue ...
I just check but I can't find that statement/error report anymore. @Koenkk  will know ...
I read that and I mad a mental note, to do and check that before I reflash my coordinator.

Koenkk, 2021-08-03:
The network route errors can usually be solved by removing some devices from power for some seconds. This will trigger routes to be rediscoverd.

pznamenskii, 2021-08-03:
what about the timeouts?
having a lot of them randomly today after flashing coordinator to the latest firmware (after stopping-starting z2m service from superviser and restarting whole HA to deal with proper backup generation).

```
Zigbee2MQTT:error 2021-08-03 22:54:13: Publish 'set' 'state' to 'curtain_room' failed: 'Error: Command 0x00158d00039cdd57/1 closuresWindowCovering.stop({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 20944 - 1 - 65 - 258 - 11 after 10000ms)'
Zigbee2MQTT:info  2021-08-03 22:54:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'curtain_room' failed: 'Error: Command 0x00158d00039cdd57/1 closuresWindowCovering.stop({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 20944 - 1 - 65 - 258 - 11 after 10000ms)'","meta":{"friendly_name":"curtain_room"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:info  2021-08-03 22:54:18: MQTT publish: topic 'zigbee2mqtt/switch_wireless_relay_kitchen_table_leds', payload '{"consumption":2.28,"energy":2.28,"interlock":null,"linkquality":153,"power":0.02,"state":"OFF","state_l1":"OFF","state_l2":"OFF","temperature":44,"voltage":226}'
Zigbee2MQTT:error 2021-08-03 22:54:22: Publish 'set' 'state' to 'switch_bathroom' failed: 'Error: Command 0x842e14fffefec098/2 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 782 - 2 - 54 - 6 - 11 after 10000ms)'
```

how do we actually repair the routers? leave-join?

Borcon, 2021-08-03:
@Koenkk 
Thanks for your help. Removing from power solved the timeout issue. 👍
I can confirm that the coordinator backup.json is not recreated when stopping or restarting Zigbee2Mqtt via the supervisor.
Only via Z2M WebGui directly. You should add this info to the changelog or fix that issue.
Many thanks for your work with Z2M. 👏

elgerg, 2021-08-04:
Well, I ended up killing everything. Removing Zigbee2MQTT and reinstalling/re pairing everything from scratch. That was painful.

closed 2021-08-04 by elgerg