Koenkk/zigbee2mqtt #8235: Malformed ZCL IAS WD Start Warning (0x00) command

neugenn, 2021-08-03:
## What happened
I'm trying to add support for a iHORN indoor siren.
I'm doing this as described here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html, by using an external converter (see the attached LH09521.js converter file).
After successfully receiving and parsing the commandStatusChangedNotification messages, I've tried to trigger the alarm on the device by publishing to: zigbee2mqtt/iHORN_siren/set the payload  '{"warning": {"mode": "burglar", "level": "medium", "strobe": true, "duration": 5}}', but nothing happened. The device responded with ZCL Default Response with Status Malformed Command (please see the attached malformed_start_warning_command.log and its corresponding Wireshark trace iHORN-malformed_start_warning_command.pcap)

## What did you expect to happen
Expected behavior: the device starts emitting audio (alarm sound) and video (led flashing) warnings. 

## How to reproduce it (minimal and precise)
This looks like a problem affecting all IAS WarningDevices.

1. Include a supported IAS Warning Device
2. Publish to its MQTT topic (zigbee2mqtt/<device_friendly_name>/set) the following payload: {"warning": {"mode": "burglar", "level": "medium", "strobe": true, "duration": 5}}'

## Debug info
Zigbee2MQTT version: latest (commit #70891eec)
Adapter hardware: CC2531
Adapter firmware version:
'{"meta":"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'

I've managed to successfully issue a IAS WD Start Warning command by using the patches available in the zip.

[malformed_start_warning_logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/6923911/malformed_start_warning_logs.zip)

Koenkk, 2021-08-03:
the current definition is indeed not correct, spec:

[image]

Fixed this now.

Also added the LH09521.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Can you check if it works correctly there?

neugenn, 2021-08-04:
Confirm working on dev branch. Thanks for fixing !

closed 2021-08-04 by neugenn

TheStigh, 2021-08-06:
@neugenn 

Does the "menu" choices from UI-> Exposes now for for you? In Dev, they do not for the siren I'm currently implementing. Changing the volume level changes the type of sound, not the volume.

neugenn, 2021-08-11:
@TheStigh 

It might have something to do with the WD's firmware maybe ?
For example, my WD (iHORN LH-09521 indoor siren) emits the same sound, no matter what the Warning mode is selected in the 
GUI (although the ZCL frame that gets to the device contains the proper value, as selected in the GUI) 

I'm currently running version 1.21.0-dev (commit #a380df4f). With this version, I can confirm (by sniffing the ZIgbee traffic) that 
the warning mode (burglar, fire, emergency etc),  siren level, strobe and warning duration are working as  expected.

Not sure about strobe level and strobe duty cycle. The corresponding bytes from the ZCL payload are being updated inline with the GUI selections, but my WD acts the same in respect to this.

TheStigh, 2021-08-11:
@neugenn 

`tz.warning` was not correct, a new will be available in Dev in days.

If you have a local converter, can you please try to add following new warning:
```
    warning: {
        key: ['warning'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
            const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
            const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
            const values = {
                mode: value.mode || 'police_panic',
                level: value.level || 'very_high',
                strobe: value.hasOwnProperty('strobe') ? value.strobe : false,
                duration: value.hasOwnProperty('duration') ? value.duration : 60,
                strobeDutyCycle: value.hasOwnProperty('strobe_duty_cycle') ? value.strobe_duty_cycle * 10 : 0,
                strobeLevel: value.hasOwnProperty('strobe_level') ? strobeLevel[value.strobe_level] : 1,
            };
            const info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
            await entity.command(
                'ssIasWd',
                'startWarning',
                {startwarninginfo: info, warningduration: values.duration,
                    strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
                utils.getOptions(meta.mapped, entity),
            );
        },
    },

```

neugenn, 2021-08-11:
@TheStigh 
> 
> `tz.warning` was not correct, a new will be available in Dev in days.
> 
> If you have a local converter, can you please try to add following new warning, test and report back ?
> 
> ```
>     warning: {
>         key: ['warning'],
>         convertSet: async (entity, key, value, meta) => {
>             const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
>             const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
>             const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
>             const values = {
>                 mode: value.mode || 'police_panic',
>                 level: value.level || 'very_high',
>                 strobe: value.hasOwnProperty('strobe') ? value.strobe : false,
>                 duration: value.hasOwnProperty('duration') ? value.duration : 60,
>                 strobeDutyCycle: value.hasOwnProperty('strobe_duty_cycle') ? value.strobe_duty_cycle * 10 : 0,
>                 strobeLevel: value.hasOwnProperty('strobe_level') ? strobeLevel[value.strobe_level] : 1,
>             };
>             const info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
>             await entity.command(
>                 'ssIasWd',
>                 'startWarning',
>                 {startwarninginfo: info, warningduration: values.duration,
>                     strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
>                 utils.getOptions(meta.mapped, entity),
>             );
>         },
>     },
> ```

This is not working for me.
Not sure why the startwarninginfo value had to be computed in 'reverse'. It doesn't seem to be inline with the specs.

TheStigh, 2021-08-11:
Hmmm, worked on two different sirens here.

May I kindly ask you to share your entire local converter, please ?

neugenn, 2021-08-11:
@TheStigh 

It's the default converter from version 1.21.0-dev (commit #a380df4f)

warning: {
        key: ['warning'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
            const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
            const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};

            const values = {
                mode: value.mode || 'emergency',
                level: value.level || 'medium',
                strobe: value.hasOwnProperty('strobe') ? value.strobe : true,
                duration: value.hasOwnProperty('duration') ? value.duration : 10,
                strobeDutyCycle: value.hasOwnProperty('strobe_duty_cycle') ? value.strobe_duty_cycle * 10 : 0,
                strobeLevel: value.hasOwnProperty('strobe_level') ? strobeLevel[value.strobe_level] : 1,
            };

            const info = (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level]);
            await entity.command(
                'ssIasWd',
                'startWarning',
                {startwarninginfo: info, warningduration: values.duration,
                    strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
                utils.getOptions(meta.mapped, entity),
            );
        },
    }

Can you, by any chance, sniff the Zigbee wireless traffic ? It would be interesting to see what is transmitted over the air.

TheStigh, 2021-08-11:
Currently I haven't got the time to do the sniffing, sorry.

Try replace your entire local warning with my code above.

TheStigh, 2021-08-11:
@neugenn ...and how did the siren behave?

neugenn, 2021-08-11:
@TheStigh 

It might be the firmware ... after all, I don't think that it's a Zigbee certified device.
The strange thing is that with the dev branch, I can control it as following:

Siren level low = no audio output. Gradually increased audio output for medium level, high level, very high level.
Strobe on or off. When strobe is on, there is a red led flashing.
Warning duration: audio and/or strobe for the required time period.

What is strange:
Audio warning is the same for all the types (panic, burglar, fire etc)
Strobe cycle and intensity: when strobe is on, the strobe behavior is the same, no matter what duty cycle and intensity value are selected.

The device itself looks pretty much like the one from the link provided by you, but it has no logo printed on it, and, when queried, it reports back model LH09521.  Didn't find anything related when googling it.

With the changes, I lost the control on the strobe (always set to no) and siren level.

Koenkk, 2021-08-12:
@neugenn can you confirm that with the current code (z2m 1.21.0 the log matches the values in the wireshark sniff). If yes, it means that the device of @TheStigh are actually not compliant to the zigbee spec and the tweak (https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files) should only be applied to these devices.