Koenkk/zigbee2mqtt #8252: Aqara ZNCZ11LM can't pair new devices.

xfzha, 2021-08-04:
## What happened
Aqara ZNCZ11LM cannot pair with new devices, but can connect to already paired devices.
## What did you expect to happen
ZNCZ11LM can pair new devices normally.
## How to reproduce it (minimal and precise)
Open ZNCZ11LM alone to pair, and then pair the new device, nothing happens with z2m.
My ZNCZ15LM can be paired normally after opening the pairing alone, And there is no information about ZNCZ11LM in <code>coordinator_backup.json</code> .
## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware:  zig-star CC2652P2
Adapter firmware version: 20210708

w3host, 2021-08-04:
Hi xfzha,

Yesterday I have tried to pairing new device (IKEA Led bulb, LED1836G9) with Zigbee2MQTT 1.21.0 and firmware 20210708 but it was not success for me as well. 
HW: Electrolama zig-a-zig-ah! (zzh!)

Aug  3 18:23:54 18087-hass-ssd npm[2978]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:23:54: Starting interview of '0x60a423fffeecab8e'
Aug  3 18:23:54 18087-hass-ssd npm[2978]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:23:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffeecab8e","ieee_address":"0x60a423fffeecab8e","status":"started"},"type":"device_interview"}'
Aug  3 18:23:54 18087-hass-ssd npm[2978]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:23:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x60a423fffeecab8e"},"type":"device_connected"}'
Aug  3 18:23:54 18087-hass-ssd npm[2978]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:23:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x60a423fffeecab8e"},"type":"pairing"}'
Aug  3 18:26:13 18087-hass-ssd npm[2978]: #033[31mZigbee2MQTT:error#033[39m 2021-08-03 18:26:13: **Failed to interview '0x60a423fffeecab8e', device has not successfully been paired**
Aug  3 18:26:13 18087-hass-ssd npm[2978]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:26:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffeecab8e","ieee_address":"0x60a423fffeecab8e","status":"failed"},"type":"device_interview"}'
Aug  3 18:26:15 18087-hass-ssd npm[2978]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:26:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x60a423fffeecab8e"},"type":"pairing"}'

I have tried it many times but same result. Already paired device working well.

So I have rolled back to previous firmware (20210120) and pairing was success again:

Aug  3 18:46:08 18087-hass-ssd npm[3167]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:46:08: Starting interview of '0x60a423fffeecab8e'
Aug  3 18:46:08 18087-hass-ssd npm[3167]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:46:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffeecab8e","ieee_address":"0x60a423fffeecab8e","status":"started"},"type":"device_interview"}'
Aug  3 18:46:08 18087-hass-ssd npm[3167]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:46:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x60a423fffeecab8e"},"type":"pairing"}'
Aug  3 18:46:56 18087-hass-ssd npm[3167]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:46:56: Successfully interviewed '0x60a423fffeecab8e', device has successfully been paired
Aug  3 18:46:56 18087-hass-ssd npm[3167]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:46:56: Device '0x60a423fffeecab8e' is supported, identified as: IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (LED1836G9)
Aug  3 18:46:56 18087-hass-ssd npm[3167]: #033[32mZigbee2MQTT:info #033[39m 2021-08-03 18:46:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":7,"description":"Controls the behavior when the device is powered on","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","previous","on"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"LED1836G9","supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x60a423fffeecab8e","ieee_address":"0x60a423fffeecab8e","status":"successful","supported":true},"type":"device_interview"}'

BR

Koenkk, 2021-08-05:
@xfzha if you ever reflashed your stick before 1.21.0 you need to repair the device which you want to pair via (see second point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0)

xfzha, 2021-08-05:
@Koenkk Thanks for the reply. When I rejoined ZNCZ11LM it worked normally.
Thanks again!

closed 2021-08-05 by Koenkk