Koenkk/zigbee2mqtt #8258: TS0203 not supported

bernd-uhl, 2021-08-05:
## What happened
The door contact TS0203 is suddenly no longer supported since 1.21.0. A new connection fails
## How to reproduce it (minimal and precise)
I see the following data in the logs
[image]
[image]

## Debug info
Zigbee2MQTT version: 1.21.0 commit: 70891eec
Coordinator type: zStack3x0
Coordinator revision: 20210320
Frontend version: 0.5.18

MarcelJ28, 2021-08-05:
Same problem here. I went back to the last version because I almost start my holiday and wanted them to work fine :-)

Koenkk, 2021-08-05:
Duplicate of #8266 and already fixed in the dev branch.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-05 by Koenkk

robert2s, 2021-08-08:
I had the same problem, and all TS0203 reappeared instantly today (2021-8-8) after I switched to add-on zigbee2mqtt **Edge** in Home Assistant. This is what I did:

1. Disable add-on zigbee2mqtt: In the Information page of zigbee2mqtt in Home Assistant, stop the add-on, and put the "start on boot" (Dutch: start bij opstarten) to Off.
2. Install add-on zigbee2mqtt **Edge**: in the add-on Store of Home Assistant, select zigbee2mqtt **Edge**, and install it.
3. Copy the configuration from zigbee2mqtt to zigbee2mqtt **Edge**:  in the Configuration page of zigbee2mqtt, copy the complete yaml text (ctrl-C). Go to the Configuration page of zigbee2mqtt **Edge**, and paste it there.
4. Start add-on zigbee2mqtt **Edge**, and switch on "start on boot" and "show in sidebar".

As soon as a new version of zigbee2mqtt is available (i.e. version 1.22.0) I recommend to reverse the above procedure, so you use the normal zigbee2mqtt add-on again.
Note: alternatively, you may decide to wait until version 1.22.0 is availble, because that will automatically solve the problems.