Koenkk/zigbee2mqtt #8259: Home Assistant V2021.8.2 has a new energy dashboard

johnjoemorgan, 2021-08-05:
## What happened
Home Assistant V2021.8.2 has a new energy dashboard that takes devices that expose power data and places those devices' power related entities into a new dashboard.

Home Assistant documentation asks users to refer to this page:

https://www.home-assistant.io/more-info/statistics/

and subsequently direct integration authors to:

https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics

## What did you expect to happen
That the Xiaomi ZNCZ02LM that exposes power and energy in the "regular" sections of Home Assistant would make the data available in the new energy dashboard as well

## How to reproduce it (minimal and precise)
Upgrade HA the latest version.  Make sure your configuration have "energy:" in the config yaml. Navigate to the energy dashboard config (/config/energy/dashboard) and follow the onscreen instructions.  Those devices with the correct configuration will appear and can be added.

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: slae.sh_cc2652rb
Adapter firmware version:

abaksa1, 2021-08-05:
My power meter device, which is Hiking TS0601_din is not being detected too.
I suppose this an issue across whole integration-supported energy metering devices.

johnjoemorgan, 2021-08-05:
**Work around**

Thank you to @carlos.vroque for pointing this out in the HA update discussion

Using the UTILIRY integration you can create kWh sensors from the W utput of the device

https://www.home-assistant.io/integrations/utility_meter/

For the Xiaomi sensor I have the following in sensor.yaml
'
  - platform: integration
    source: sensor.display_fridge_small_power
    name: Small Display Fridge Power Manual
    unit_prefix: k
    unit: kWh
    round: 2
'

I have similar sensors for Sonoff and Tuya V2 devices as well

McGiverGim, 2021-08-05:
In theory only need is to add this attributes to the sensor entities:
- `state_class: measurement` for any sensor that we want to store in long term
- ` last_reset` timestamp if the sensor is a value that continuously grow, like the energy sensors.

Koenkk, 2021-08-05:
Already supported in 1.21.0 (#8187).

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-08-05 by Koenkk