Koenkk/zigbee2mqtt #8262: TuYa TS004F action exposes not as expected

Kilkenny-82, 2021-08-05:
Not sure if it should go in New device support, but recently after I updated Zigbee2MQTT to the latest version my TS004F was supported. I bought 2 of these, one is described as a TS0044 and was supported right away, the other ordered some time later added as TS004F. They both look very much the same, only the actions coming from the switches are very different. The TS0044 has for each of the 4-buttons a Press, Double Press and Hold (total of 12 actions). The TS004F has on, off and brightness step and move up & down. 
Shouldn't they work the same, since they are practically the same? I would have expected something like this:
`const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: 'TuYa',
    description: 'Wireless switch with 4 buttons',
    fromZigbee: [fz.tuya_on_off_action, fz.battery],
    toZigbee: [],
    exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
            '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold'])],
};`

Koenkk, 2021-08-05:
While these devices may look very similar from the outside, the software on these device really differs.
- TS0044: only sends non standard zigbee commands
- TS004F: sends native zigbee commands (e.g. turn on /off command). This is what you see in the actions.

closed 2021-08-05 by Koenkk

ARafaelSF, 2021-10-05:
@Koenkk 

With Tuya Hub It´s possible change mode, to device works like dimmer ou trigger.

If I made a sniff, do you think it is possible to add this n option?

[image]