Koenkk/zigbee2mqtt #8266: Removing device from blocklist doesn't work

vmartinv, 2021-08-05:
## What happened

When I remove a device from the blocklist and submit the changes the device reappears in the blocklist. This was already reported in https://github.com/Koenkk/zigbee2mqtt/issues/6563

## What did you expect to happen
The device should have been removed from the blocklist.

## How to reproduce it (minimal and precise)
**WARNING: You won't be able to connect to the device anymore!**
1. Go to Settings->Blocklist
2. Add a device to the blocklist.
3. Press '-' to remove the device from the blocklist and press submit.

[image]

## Debug info
Zigbee2MQTT version: 1.20.0 commit: 1.20.0, Frontend version 0.4.29
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127

Koenkk, 2021-08-07:
Thanks, fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-07 by Koenkk

vmartinv, 2021-08-07:
Many thanks to you @Koenkk !

danymexi, 2023-07-28:
Hello, I still have the problem. When I click on the - the device is not removed.

How can I solve the issue?

Thanks