Koenkk/zigbee2mqtt #8267: AWOX 33946 LED bulb recognised as different one

GrabkaMan, 2021-08-05:
Hey, I've bought LED GU10 bulbs from Awox. 
Here is the polish link:
https://www.castorama.pl/zarowka-led-awox-gu10-345-lm-2700-6500-k-dim-id-1134896.html 

It gets discovered as different device: 
33951/33948 https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948

Which is not completely true - My model has the option to change colors as well. 
I've tried adding the device with the documentation, but this doesn't work - the device cannot be found in any way:
I've taken the sample light js file and changed it a bit with this values:

```
    zigbeeModel: ['TLSR82xx'],
    model: '33951/33948', //Also tried 33946 here
    vendor: 'AwoX',
    description: 'Awox bulb NEW!',

```
Next I've changed the configuration file like so:

```
 "external_converters": [
    - 33946.js
  ],
  "devices": [
    "devices.yaml"
  ],
```
But this didn't change a thing. 

Database entry:
`{"id":36,"type":"Router","ieeeAddr":"0xa4c1383ba5cef6d2","nwkAddr":30100,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[6,25],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"hwVersion":0,"swBuildId":"0122052017"}},"genLevelCtrl":{"attributes":{"currentLevel":217}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":49152,"epId":3,"devId":258,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1628194770914}`

How can I make this device working correctly?

Koenkk, 2021-08-06:
This should already be fixed in z2m 1.21.0.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

GrabkaMan, 2021-08-06:
Thank you for your reply. 
In fact I was in 1.18.1. 
So I've added the repo, installed - copied old config into new version - but first problem - **502: Bad Gateway** :|

Then I thought - ok, there should be mistake somewhere - so after investigation I've changed the old "data_path" from /share/zigbee2mqtt to /data/zigbee2mqtt - leaving all other settings untouched. 

But then there is a porblem of data - I kinda lost all my devices after the upgrade. 
But that is not a problem - I still have the old version, right? So yes, in "/share/zigbee2mqtt" I still have my old version - so I wanted to copy content to "/data/zigbee2mqtt". But I don't have that folder... 
My /data/ didn't have any zigbee2mqtt folder - so I've created it. Then copied the content of share there. 
Still don't have my old devices - but at least new version is working :D

Any ideas on how to copy old devices to new version?

MGGonsior, 2021-08-06:
On zigbee2mqtt ver. 1.21.0-1, GU10 RBGW AWOX bulb is detected as 33951/33948 (https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948).
I can control brightness and turn it on/off.

GrabkaMan, 2021-08-07:
Thats correct, but this bulb is RGBW, so controlling colors would be a nice thing as well

Koenkk, 2021-08-07:
@GrabkaMan do I understand correctly that the bulb is now correctly recognised? If yes please close this issue.

MGGonsior, 2021-08-07:
The bulb isn't recognized correctly, but its possible to control brightness and turn it on/off. it isn't possible to change the colors.

GrabkaMan, 2021-08-09:
Hey all, 

I did some tests and the thing is that this device gets recognized as 33943 model (and it is 33946 - that's what is stated on the bulb itself)
And now I'm able to do a bit more with it. What works:
- on/off
- colors
- brightness
- color_temp 
- color_temp_startup

What is not working:
- effects (they are there, but : 
1. 'okay' effect makes the bulb green and its hard to change the color at all. 
2. Breathing effect makes the bulb flash with light 

Also sometimes the color is changed slowly - like a smooth transition, and sometimes its not. I don't know what is the reason of that.

As a side info: I can't use external_converters to change the device type to 33946.

GrabkaMan, 2021-08-09:
```
Info Successfully interviewed '0xa4c1382c1aed8310', device has successfully been paired
Info Device '0xa4c1382c1aed8310' is supported, identified as: AwoX LED RGB & brightness (33943)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"LED RGB & brightness","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"33943","supports_ota":false,"vendor":"AwoX"},"friendly_name":"0xa4c1382c1aed8310","ieee_address":"0xa4c1382c1aed8310","status":"successful","supported":true},"type":"device_interview"}'
Info Configuring '0xa4c1382c1aed8310'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"LED RGB & brightness","friendly_name":"0xa4c1382c1aed8310","model":"33943","supported":true,"vendor":"AwoX"},"type":"pairing"}'
Info MQTT publish: topic 'homeassistant/light/0xa4c1382c1aed8310/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0xa4c1382c1aed8310/set","device":{"identifiers":["zigbee2mqtt_0xa4c1382c1aed8310"],"manufacturer":"AwoX","model":"LED RGB & brightness (33943)","name":"0xa4c1382c1aed8310","sw_version":"Zigbee2MQTT 1.21.0"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/0xa4c1382c1aed8310","max_mireds":500,"min_mireds":150,"name":"0xa4c1382c1aed8310","schema":"json","state_topic":"zigbee2mqtt/0xa4c1382c1aed8310","supported_color_modes":["xy","color_temp"],"unique_id":"0xa4c1382c1aed8310_light_zigbee2mqtt"}'
Info MQTT publish: topic 'homeassistant/sensor/0xa4c1382c1aed8310/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1382c1aed8310"],"manufacturer":"AwoX","model":"LED RGB & brightness (33943)","name":"0xa4c1382c1aed8310","sw_version":"Zigbee2MQTT 1.21.0"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c1382c1aed8310","name":"0xa4c1382c1aed8310 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1382c1aed8310","unique_id":"0xa4c1382c1aed8310_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c1382c1aed8310', payload '{"color_mode":"xy","linkquality":255,"state":null}'
Info Successfully configured '0xa4c1382c1aed8310'
```

This is the configuration information of the device.

Koenkk, 2021-08-10:
@GrabkaMan I've changed the `model` to `33943/33946` now (this is only a cosmetic thing).

The implementation of effects differs per vendor, what you see is what is defined in the Zigbee spec, the actual effects may differ per device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-09-17 by github-actions[bot]

Koenkk, 2022-08-01:
@vavarm please provide the data/database.db entry of your device.

vavarm, 2022-08-01:
Please find below the content of data/database.db
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c82ac0","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Router","ieeeAddr":"0xa4c138317769ff61","nwkAddr":43389,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1659374997022,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"Group","groupID":1,"members":[{"deviceIeeeAddr":"0xa4c138317769ff61","endpointID":1}],"meta":{}}
```

Koenkk, 2022-08-02:
@vavarm should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

vavarm, 2022-08-02:
> @vavarm should be fixed now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thank you. It works fine now

MarkMikkelsen83, 2023-08-12:
I got this problem still...
33951/33948 its a 33943/33944/33946

Koenkk, 2023-08-13:
@MarkMikkelsen83 please provide the data/database.db entry of your device