Koenkk/zigbee2mqtt #8270: Schneider Electric "LK FUGA wiser wireless battery 4 button switch"

taugusti, 2021-08-06:
## What happened
I have tried to pair a "LK FUGA wiser wireless battery 4 button switch" to zigbee2mqtt. 
It is paired and interviewed correctly but reported as unknown device
The log entry after pairing says "Zigbee model 'FLS/AIRLINK/4' and manufacturer name 'Schneider Electric'"

## What did you expect to happen
I expected it to be detected as this device: https://www.zigbee2mqtt.io/devices/550D6001.html
The zigbee model FLS/AIRLINK/4 is the same as described for 550D6001. 

It looks the same as on the picture and description except I bought it without the plastic buttons (I need to engrave blank buttons).
I assumed it is the same device since schneider does not describe about two different battery buttons.

## How to reproduce it (minimal and precise)
3x click on button 1 (top left) makes it pair with zigbee2mqtt
My device says 550D0001 on the front - not 6001 as in the link. 

I tried to follow the "how to support new devices" but when I type in a .js file name in the configuration, zigbee2mqtt fails during restart: https://pastebin.com/6yFtPa4w
I tried to copy the 4-button switch config from the vendor.js file but it did not change the looping restart problem

Zigbee2mqtt version: 1.18.1 (on home assistant 2021.7.4 with zigbee2mqtt addon version 1.18.1-1)
Adapter hardware: CC26X2R1
Adapter firmware version: 20210120 (when I updated to 0708 the device did not pair at all)

I hope you can help
/th

Koenkk, 2021-08-06:
Device is already supported in z2m 1.21.0: https://github.com/Koenkk/zigbee-herdsman-converters/blob/67918aaaa314c02dc49d7429c0e0af728a5bc9ac/devices/schneider_electric.js#L293

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-08-06 by Koenkk

taugusti, 2021-08-08:
Thanks, I was not aware of the moved repository and related change of hass configuration.
It works now