Koenkk/zigbee2mqtt #8271: network slow down after time

phenix1990, 2021-08-06:
- If applicable, provide steps how to reproduce the problem.
- Provide the herdsman debug logging: https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

## What happened
Since 2 weeks network is slowing down after some time. Single command can take multiple seconds to react. Problem started without any changes in V1.17 (dont remember for sure) and continues with newest coordinater flashed and z2m V1.21 dev and master branch

## What did you expect to happen
Fast performance regardless of running time

## How to reproduce it (minimal and precise)
After restart high perfomance, then slow down in short time (maybe 10 minutes)

## Debug info
Zigbee2MQTT version: 1.21 dev
Adapter hardware: CC2652RB
Adapter firmware version: coordinator Z-Stack 3.x.0 20210708 

I guess it maybe has to do with my Blitzwolf SPH-13 devices. they spam the network and getting timeouts. But i dont understand how this issue started because i use them for many months now and for long time all was working fine

<pre>
Aug 06 05:40:24 openHABianDevice npm[5835]: Zigbee2MQTT:error 2021-08-06 05:40:24: Publish 'set' 'state' to 'PlaystationPower' failed: 'Error: Command 0x588e81fffed39555/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 27577 - 1 - 6 - 6 - 11 after 10000ms)'
Aug 06 05:40:24 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:40:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'PlaystationPower' failed: 'Error: Command 0x588e81fffed39555/1 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 27577 - 1 - 6 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"PlaystationPower"},"type":"zigbee_publish_error"}'
Aug 06 05:43:02 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:02: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":69,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:02 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:02: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":66,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:02 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:02: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":69,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:02 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:02: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":69,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:02 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:02: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":69,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:03 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:03: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":69,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:03 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:03: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":57,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:03 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:03: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":57,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:03 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:03: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":57,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:03 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:03: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":57,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:04 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:04: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardwareVersion":1,"ieeeAddr":"0x588e81fffed39555","manufacturerID":4098,"manufacturerName":"_TZ3000_g5xawfcq","model":"TS0121_plug","networkAddress":27577,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":5.2,"linkquality":63,"power":0,"state":"OFF","voltage":232}'
Aug 06 05:43:04 openHABianDevice npm[5835]: Zigbee2MQTT:info  2021-08-06 05:43:04: MQTT publish: topic 'zigbee2mqtt/PlaystationPower', payload '{"current":0,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"PlaystationPower","hardware [... cut ...] e\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 27577 - 1 - 27 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"PlaystationPower"},"type":"zigbee_publish_error"}'
</pre>

In the frontend i tried to change the reporting to 60 (seconds?) and only set reporting for onOff and active power
[image]

I hope someone can help because my wife gets angry about not working lights :smile:

pedrolamas, 2021-09-01:
I've been noticing something similar to this myself, mostly on the battery operated devices but I think I also see this on the others too!

My Aqara Xiaomi MCCGQ11LM door/window sensor always worked perfectly and for a while now I've been noticing quite a delay on the events appearing on the logs...

The weirdest bit is that if I take one and using a magnet and open and close a few times, I can actually heard the "relay" inside of it, but nothing shows on the debug log, but then I press the button on the sensor and get all the events as if they were buffered!

@Koenkk did something change over the last couple of versions that could have caused this? I'm using the latest (1.21.1, launched today), but I also noticed this behavior in the previous one...

pedrolamas, 2021-09-01:
Further information, my zzh has been running CC2652R_coordinator_20210708.hex since it was launched back in August, I just tried reflashing the stick and I can still reproduce this issue, **even with the door sensor 20cm away from the coordinator stick!**

Koenkk, 2021-09-01:
@pedrolamas cannot relate this to a change yet. 

Since you have a nice way to reproduce the issue (with the magnet), can you confirm that the sensor immediately sends the `attributeReport` once you trigger the contact? In this way we can find out if the delay is at the z2m or device side. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

pedrolamas, 2021-09-01:
I can confirm right now that on the logs I do not see the attributeReport until I actually press the button on the sensor (that makes it send a bunch of information), but let me get my old CC2531 (need to find it first!) and I'll check!

pedrolamas, 2021-09-01:
@Koenkk, I've confirmed the issue!!

My CC2531 connect to my Windows machine, with ZBOSS and WireShark, I can see the events happening in real time, but the coordinator does nothing, no messages appear (I even enabled herdsman debug and nothing shows up)...

Does this point to some issue with the coordinator firmware?

Koenkk, 2021-09-02:
> Does this point to some issue with the coordinator firmware?

The delay could still be in one of the hops, could you share the sniff +  network key?

pedrolamas, 2021-09-02:
Here is the trace, I'll send the key via Discord (or other method if you prefer)

[temp.zip](https://github.com/Koenkk/zigbee2mqtt/files/7100044/temp.zip)

lassieee, 2021-09-14:
I have the exact same issue. Using zigbee2mqtt:1.21.1 in docker with a CC2652R adapter and Z-Stack 3.x.0 20210708. All Xiaomi door sensors have a delay of 5-10 seconds. When sniffing I can see the traffic they produce.