Koenkk/zigbee2mqtt #8272: last_reset not updating for some energy sensors or energy value not correct

McGiverGim, 2021-08-06:
## What happened

Home Assistant has a new Energy Dashboard. It was supported in PR https://github.com/Koenkk/zigbee2mqtt/pull/8187 
The energy sensors need to have a `last_reset` value but something seems not to be working.

This is a LLKZMK11LM Xiaomi relay. The last reset seems to be a fixed value:

[image]

But the sensor value is resetting itself, I don't know under which circumstances, from time to time. For example here you can see two resets in the latest 24 hours:

[image]

The result of this is that the entity appears with a "negative" value in the new dashboard:

[image]

The most strange thing is that I have other two similar relays, and they seem not to be resetting never. The three of them work perfectly (the only difference is that the other two have the interlock enabled but I doubt it matter). So I'm not sure if this is a hardware problem (despite it seems to be working ok), or some bug in the software.

## What did you expect to happen

The last reset value to be correct or the energy value of the sensor not resetting itself. 

Maybe in my case is some kind of problem in the hardware, but I suppose that at some point the energy values will reset, maybe when we reconnect the device for example and we press the reset button for 10 seconds, so I suppose we need to detect it and adjust the `last_reset` value.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

McGiverGim, 2021-08-06:
Another difference with the problematic device, is that I define it as light and not as a switch:
```yaml
'0x00158d00044bda5b':
  friendly_name: rele_cocina
  homeassistant:
    switch_l1:
      type: light
      object_id: light1
    switch_l2:
      type: light
      object_id: light2
'0x00158d00044adbcc':
  friendly_name: rele_comedor_mesa
'0x00158d0003f453e3':
  friendly_name: rele_comedor_centro
```

Koenkk, 2021-08-06:
The first step is to figure out how to reset it, then we should check if we can capture this somehow.

For now the assumption was that the `energy` value could not be reset.

McGiverGim, 2021-08-06:
I suppose that if the energy value is smaller than the last value, then reset the date.

McGiverGim, 2021-08-12:
I've tried to "reset" the relay, to see if this fixes the issue with the energy going back to 0.6kWh randomly (I will confirm later if this makes some change).

This is the graph of the latest hours, several return to 0.6 kWh and at the end, it goes to 0.1 kWh after the reset and rejoining the network.

[image]

Here is some log of the new rejoin:
```log
info  2021-08-12 12:36:12: Device 'rele_cocina' joined
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"rele_cocina","ieee_address":"0x00158d00044bda5b"},"type":"device_joined"}'
info  2021-08-12 12:36:12: Starting interview of 'rele_cocina'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"rele_cocina","ieee_address":"0x00158d00044bda5b","status":"started"},"type":"device_interview"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"rele_cocina"},"type":"device_connected"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"rele_cocina"},"type":"pairing"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"last_seen":"2021-08-12T12:36:12+02:00"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"last_seen":"2021-08-12T12:36:12+02:00"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"rele_cocina","ieee_address":"0x00158d00044bda5b"},"type":"device_announce"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"rele_cocina"},"type":"device_announced"}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":75,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":75,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:12: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:12+02:00","linkquality":78,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:13: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"energy":null,"interlock":null,"last_seen":"2021-08-12T12:36:13+02:00","linkquality":78,"power":null,"state_l1":null,"state_l2":null,"temperature":null,"voltage":null}'
info  2021-08-12 12:36:13: Successfully interviewed 'rele_cocina', device has successfully been paired
info  2021-08-12 12:36:13: Device 'rele_cocina' is supported, identified as: Xiaomi Aqara wireless relay controller (LLKZMK11LM)
info  2021-08-12 12:36:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara wireless relay controller","exposes":[{"access":5,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":3,"description":"Enab [... cut ...]
Maybe my unit is a defective one, but is clear that after a reset and rejoin to the network, the energy value has been reset and it will be a good idea to update the `last_reset` value. I'm not too sure what is the good condition:
1. A rejoin to the network
2. An Energy value of 0.1 (or 0.0, the relay was on consuming energy, so maybe it can receive a 0.0 too) 
3. An Energy value smaller than previous known value
4. More than one of the upper conditions

I think that an AND with the options 2 and 3 will be a safe one and it will work.

nattgris, 2021-08-17:
Consider that metered measurements are AFAICT not necessarily monotonically increasing (the fact that HA shows a negative bar for a decreasing value should be a strong indicator that it's meant to be possibly decreasing). A practical example would be an energy meter in the case of local electricity production.

Also, seeing a value close or equal to zero is not robust because communication could be down while the meter resets and consumption is still active. The 0.01 in your case could just as well be 10.0 in another situation with a bit longer delay and higher consumption.

All meters will eventually reset at the source because the report has limited amount of bits available to represent the growing value. "64 bits ought to be enough for everyone" though...

IF we knew that a particular device is strictly increasing (like a mm rain gauge) AND we know what the last value was before the reset (implicitly through a known fixed wrap-around value, or by the last hopefully recent report before a device reset), THEN we can either undo the reset/unwrap and report an accumulated value, or notify HA via the last_reset mechanism that the metered value has not in fact decreased. Otherwise we're in the dark.

If we need to make a guess based on tracking the actual value, I'd say that logic would be better implemented inside HA to all integrations' benefit.

The last_reset mechanism is kind of flawed in that HA has no idea in the current model at what value the reset actually happened, so HA is bound to accumulate an error compared to the actual meter.

McGiverGim, 2021-08-18:
Hi @nattgris 

I don't say that we need to do an "automatic" reset for all the devices. As you say some of them can have negative consumption. But some others not, and I think we need to fix at least the case when we reset the device.

My device seems buggy, because it resets in a random way, but for any device, if we change the network key, for example, we need to rejoin all the devices, resetting them and causing it to go to zero (at least with xiaomi sensors but I suspect that this will happen with any other brand), producing a bad statistic. This statistics are stored now in long term so I think is a good idea to set the last_reset value in a good way.

I don't know if add some parameter to the device directly in the code or add a parameter in the `devices.xml` similar to `reset_zero_join: true` or similar.

McGiverGim, 2021-08-18:
It seems that Home Assistant is working on fixing that... 

https://github.com/home-assistant/core/pull/54831

If I'm not wrong, a new `state_class: total_increasing` has been added. Is similar to the old `measurement` but with automatic `last_reset`.

closed 2021-09-25 by github-actions[bot]