Koenkk/zigbee2mqtt #8275: Aqara Opple 6btn switch double click and hold functions don't work

tomasko25, 2021-08-06:
## What happened
I am running zigbee2mqtt on Rpi 3b+ with ConBee II USB gateway and I paired Xiaomi Aqara Opple switch 3 bands. Single clicks work, hold&release works on btns 4-6, double and triple click don't work. After double click for example btn1, there is payload like this: "button_1_single". While holding btn1 or 2, payload is the same. After tripple click, no reaction.
I tried running zigbee2mqtt add-on in Home Assistant on Rpi with ConBee II and everything worked (double also tripple click).

## What did you expect to happen
Double and triple click working on all btns, hold&release working on btns 1 & 2.

## How to reproduce it (minimal and precise)
Single clicks work, hold&release works on btns 4-6, double and triple click don't work. After double click for example btn1 there is payload like this: "button_1_single". While holding btn1 or 2, payload is the same. After tripple click, no reaction.

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: Conbee II
Adapter firmware version: ?

fluffymadness, 2021-08-09:
Try unbinding and rebinding it to the cordinator again in the binding tab.
I had issues like these on mine too.

tomasko25, 2021-08-11:
> Try unbinding and rebinding it to the cordinator again in the binding tab.
> I had issues like these on mine too.

Are there any special unbinding steps?

closed 2021-09-19 by github-actions[bot]