Koenkk/zigbee2mqtt #8276: LELLKI WP33-EU not supported

bernd-uhl, 2021-08-06:
## What happened
Hello I would like to teach a LELLKI WP33-EU to Z2M but this does not work:
[https://www.zigbee2mqtt.io/devices/WP33-EU.html](url)
I've been trying this for months now in the hope that this will work out at some point.
Thank you very much for the help.
## How to reproduce it (minimal and precise)
get the following information in the logs:
[image]
[image]
## Debug info
Zigbee2MQTT version: 1.21.0-dev commit: 10615e29
Coordinator Typ: zStack3x0
Coordinator Version: 20210320
Frontend Version: 0.5.19

bernd-uhl, 2021-08-06:
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/WP33-EU.md

Koenkk, 2021-08-07:
This version is already supported in the latest dev branch (second fingerprint of https://github.com/Koenkk/zigbee-herdsman-converters/blob/c629717fa7cabde128371253df09befaaf0a1585/devices/lellki.js#L11)

closed 2021-08-07 by Koenkk