Koenkk/zigbee2mqtt #8287: Add sensor.last_seen when homeassistant_legacy_entity_attributes is false

Enzokot, 2021-08-07:
## Is your feature request related to a problem? Please describe
If I set `homeassistant_legacy_entity_attributes` to false, then the `last_seen` and `device` attributes will be removed from all sensors,  that is expected.

## Describe the solution you'd like
Please, add `sensor.ieeeAddr_last_seen` for all devices when `homeassistant_legacy_entity_attributes` is `false` and advanced option `last_seen` is `on`

## Describe alternatives you've considered
I don't know any alternative ways to solve the problem.

## Additional context
-

andrew-d, 2021-09-07:
Not stale

Enzokot, 2021-09-07:
Keep alive

Enzokot, 2021-10-08:
Keep alive

andrew-d, 2021-10-08:
Speaking personally, I have a notification that goes out periodically that tells me what devices are unreachable, and having the 'last seen' date is helpful to know if it's temporary (i.e. I ignore devices that are only "unavailable" for short periods).

Enzokot, 2021-12-11:
@Koenkk when I open `mqtt` integration frontend page this error appears: 
```
Logger: frontend.js.latest.202111090
Source: components/system_log/__init__.py:190
First occurred: 18:24:34 (1 occurrences)
Last logged: 18:24:34

http://homeassistant.local:8123/frontend_latest/20a1269e.js:1:351 Uncaught RangeError: Value need to be finite number for Intl.RelativeTimeFormat.prototype.format()
```
My configuration contains `last_seen: epoch` and this add `sensor.*_last_seen` with `device_class: timestamp` but `timestamp` is only for: `Datetime object or timestamp string (ISO 8601)`

Koenkk, 2021-12-14:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-01 by Enzokot

jschwalbe, 2023-12-31:
Can anyone explain how to make this show up in HA?

It's showing up correctly in MQTT:
[image]

But HA won't let me see the last_seen value:
[image]

Any thoughts? Thanks!