Koenkk/zigbee2mqtt #8288: No converter available for 'linkquality' _TYZB01_qeqvmvti (TS0011)

elgerg, 2021-08-07:
Hi,
I have a couple of these TS0011 devices. They when turned on or off report no converter for link quality.

## What happened
[image]

However sometimes it must report properly as it does seem to work:
[image]

## What did you expect to happen
I expected the link quality to be updated

## How to reproduce it (minimal and precise)
Turn the switch on or off or wait for it to announce itself on the network

## Debug info
Zigbee2MQTT version:1.21.0
Adapter hardware: CC2531
Adapter firmware version: 3.0

Koenkk, 2021-08-07:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

closed 2021-09-15 by github-actions[bot]

xychix, 2022-11-07:
[no action needed, purely for future reference]

I was about to post logs for this issue when I found my issue.
getting a json from a dimmer and sending that to a lamp / other dimmer failed as it had the linkquality field in the json.

using a node-red change object to delete that msg.payload.linkquality field before sending the json out solved this.