Koenkk/zigbee2mqtt #8292: No converter available for 'transition' (10) from a Ubisys D1

Danit2, 2021-08-07:
## What happened

I have an automation that should switch off a lamp with a 'transition' time of 10 seconds.
https://hastebin.com/icewovuyux.yaml
The lamp switches off and the 'transition' also works. But I get the following error message in my logs:
https://hastebin.com/udelecorog.apache 

## What did you expect to happen

No Error

## How to reproduce it (minimal and precise)

Whenever the automation is started

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: 20210120

Danit2, 2021-08-07:
There's a similar topic here:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/2842#issue-953329449

Koenkk, 2021-08-09:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-09 by Koenkk