Koenkk/zigbee2mqtt #8295: Device with Zigbee model 'TS0203' and manufacturer name '_TZ3000_f1hmoyj4' is NOT supported

bambo86, 2021-08-07:
## What happened

After the z2m update (Current version: 1.21.0-1 ), some of my door open sensors show as unsupported.
It is the TS0203 model
and it is in the 'zigbee-herdsman-converters' database in the tuya.js file that zigbee2mqtt uses.
So it should work

I don't know what's going on 🤔

When I have paired these sensors for the first time (about a month ago), they were also unsupported, and then I've paired they one more time and they worked normally as supported.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.21.0-1 
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_ConBeeII_0x26680700.bin.GCF (30-Nov-2020 18:03)

robert2s, 2021-08-08:
Is a duplicate of #8266 and fixed in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Koenkk, 2021-08-09:
Already fixed in the dev branch indeed.

closed 2021-08-09 by Koenkk