Koenkk/zigbee2mqtt #8296: Hue fugato single variant support

sygys, 2021-08-07:
I just started using z2mqtt and added allot of my devices to it. What didnt work out was adding the fugato spots. When looking at the device support page is saw that almost all fugato spots are supported except the single version: 

[image]

I read the page for adding a converter but i cant seem to make a start on this. I just dont know where to start and what code to use in the .js file. And i guess like most of the fugatos are allready supported it should be great if the single version was too.

Could some one please provide me with detailed info on how to do this?

iEEE address: 0x00178801068dedbb (0x0FF0)
Model no.: 5063131P7
Log info: Device '0x00178801068dedbb' with Zigbee model '5063131P7' and manufacturer name 'Philips' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Thanks in advance.

sygys, 2021-08-07:
It keeps trying but this is the error it keeps giving:

https://pastebin.com/MupnHmwZ

sygys, 2021-08-07:
For some reason when i edited the configuration.yaml file in config/z2mqtt it kept resetting to the old values when i restarted Z2mqtt. Zo i added the external_converter to the configuration tab in the addon it self but when i do that z2mqtt wont start anymore and i end up having the 502 error bad gateway.

sygys, 2021-08-08:
It's seems that I need to install herdsmen something not explained in the docs. If you tell people who are new to this to first try to add it themselves it would be great if everything is explained. Someone like me doesn't even know how to install herdsmen on HA on a raspi... Is this a custom component? Or where do I put the files?

sygys, 2021-08-08:
Just wanted everyone to know i was running an old version of Zigbee2mqtt. in the addon store in ha i had an old repository which has been changed since. i now run 1.21 and it all works well. i will post the code that did the job for me adding everything i needed. Maybe in a future release this data can be used. Its for the white and black version of the 1 spot fugato Hue

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

module.exports = [
    {
            zigbeeModel: ['5063131P7'],
            model: '5063131P7',
            vendor: 'Philips',
            description: 'Hue Bluetooth white & color ambiance spot Fugato white (1 spots)',
            meta: {turnsOffAtBrightness1: true},
            extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
            ota: ota.zigbeeOTA,
        },

    {
            zigbeeModel: ['5063130P7'],
            model: '5063130P7',
            vendor: 'Philips',
            description: 'Hue Bluetooth white & color ambiance spot Fugato black (1 spots)',
            meta: {turnsOffAtBrightness1: true},
            extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
            ota: ota.zigbeeOTA,
        },

];
```

Koenkk, 2021-08-09:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-09 by Koenkk