Koenkk/zigbee2mqtt #8297: ZNCLDJ12LM (Aqara B1 curtain motor) OTA update not working

Chriserus, 2021-08-07:
## What happened

OTA update for ZNCLDJ12LM (Aqara B1 curtain motor) is not working. I get "Failed to check if update available for 'Curtain' (Cannot read property 'ID' of undefined)", the same goes for manual command for checking for update and trying to directly update the device "Update of 'Curtain' failed (Cannot read property 'ID' of undefined)" (as explained on OTA wiki page https://www.zigbee2mqtt.io/information/ota_updates.html#manually-check-if-an-update-is-available and https://www.zigbee2mqtt.io/information/ota_updates.html#update-to-latest-firmware ). I am running the latest z2mqtt version, my curtain motor: Build date: 20190329, Firmware version: 2019www. 

## What did you expect to happen

An update to occur / is update available check success for Aqara B1 curtain motor.

## How to reproduce it (minimal and precise)

I hope it's replicable with the above mentioned device and newest z2mqtt version.

## Debug info
Zigbee2MQTT version: 1.21.0-1
Adapter hardware: CC2531

Koenkk, 2021-08-09:
Please provide the debug log of this.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

Chriserus, 2021-08-09:
Here's a complete log from manual update check: https://pastebin.com/QqjZ2MZ0
And here's a log from trying to manually update the device: https://pastebin.com/DXWm7jXQ

Koenkk, 2021-08-10:
Added some more logging, can you update to the latest dev branch and provide the debug logging again? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Chriserus, 2021-08-11:
Sorry for the delay, here are the logs:
OTA check https://pastebin.com/E3gB68a2
OTA update https://pastebin.com/pRUG4Eur
I hope this helps :)

Koenkk, 2021-08-12:
I've made the error more human friendly now, but the root cause is that your device does not have an OTA endpoint. Please share your data/database.db entry so I can confirm this.

Chriserus, 2021-08-12:
Here is the database.db entry of my device: https://pastebin.com/ckeJiHKw

Chriserus, 2021-08-12:
If that's the case, that my device does not support OTA, we should also update the wiki page for the device, to not mislead people (if that's the case, that it shows like it is possible/supported but it is not)

Koenkk, 2021-08-13:
Your device indeed does not have the OTA cluster, what you can try is adding it and see if it works:
- stop z2m
- add `25` to the `outClusterList` of your ZNCLDJ12LM 
- start z2m and attempt the ota update, does this work?
-

Chriserus, 2021-08-13:
I have followed your instructions and indeed, the update has started! Here are the logs from requesting an update for the device after adding 25 to outClusterList: https://pastebin.com/F24aBBQG

It takes some time, I'll let you know if it succeeds.

Chriserus, 2021-08-13:
It seems like it succeeded, but I don't see a version change, see finishing logs here https://pastebin.com/MYLCPWF6

Koenkk, 2021-08-15:
can you try repairing the device?

z-x-f, 2021-08-17:
delete it and rejoin.

Chriserus, 2021-08-18:
Repairing, like, repair a car, why didn't I think about pairing the device again? :D 
I did it, it then showed that an update is available, I did an update, it was successfull, but the version is still old, nothing changed.

closed 2021-09-25 by github-actions[bot]

jeroenvdd, 2022-01-20:
This motor is still not getting the correct OTA info, update via xiami hub is far more recent

Hessenpower01, 2022-08-19:
Same here, mine ist stuck on version 2019www. dated 20200803

tried adding "25" to outclusterlist, no effect

Rig820, 2022-09-23:
How do you add to the outClusterLIst?

Jarikk, 2022-10-01:
The same question: How I can manually add OTA cluster?
Mine is also stuck at the :
Firmware build date 20190329
Firmware version 2019www.

I added it to the Mi account via Xiaomi Gateway 3 - but there are no any firmware updates available in the Mi Home application.

Hessenpower01, 2022-10-01:
In here: /config/zigbee2mqtt/database.db

new-kirte, 2022-10-06:
@Koenkk adding 25 to outClusterLIst works fine - firmware updates appear and getting applied. Is if possible to add this endpoint automatically for Aqara B1 curtains?

Jarikk, 2022-10-06:
@new-kirte Are you sure that new firmware has actually been downloaded to your motor? I added 25 to the clusters list, ran available OTA update for my curtains and after an hour of updating the device firmware version remains the same as it was before updating.

new-kirte, 2022-10-06:
> @new-kirte Are you sure that new firmware has actually been downloaded to your motor? I added 25 to the clusters list, ran available OTA update for my curtains and after an hour of updating the device firmware version remains the same as it was before updating.

Yes it does. The version number is changing