Koenkk/zigbee2mqtt #8302: MOEs ZigBee Dimmer Module (AliExpress) 1-ch support

gert3d, 2021-08-08:
Hi there,
I have no clue what to do with the information on:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/eba65cddc4f4abe5f1e7661e8145ed3bed2c8b6b
to add support for a 1-ch moes module (I am using Domoticz).
Can anyone direct me to some explanation of steps to be taken?
thanks
[image]

robertalexa, 2021-08-08:
The device should be supported. You shouldn't need to do anything more than pairing. Support is available in z2m 1.21.0. Make sure you are on the latest version
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0
[image]

gert3d, 2021-08-09:
Thanks. 
However ... next problem: I cannot find information how to update zigbee2mqtt running with Domoticz.
Is it as simple as swapping for new file(s)?
Probably not the right place to ask this, sorry for that.

robertalexa, 2021-08-09:
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#6-for-later-update-zigbee2mqtt-to-the-latest-version

gert3d, 2021-08-09:
Thanks, I'll give it a try.

closed 2021-09-16 by github-actions[bot]