Koenkk/zigbee2mqtt #8303: zigbee2mqtt cannot connect to mqtt server in container environment

fboerman, 2021-08-08:
hi,

I have spend hours on this but I cant get it to work :(
I am trying to run zigbee2mqtt on a raspberry pi with podman. This should be a dropin replacement for docker. So the flags you see are the same as for docker (in case a reader doesnt know podman).

To make sure podman was not the problem I have tried to debug as much as possible.

What I do:
create a pod and two containers within them. one for mosquitto one for zigbee2mqtt.
```
podman pod create --name=hass --share net
podman create --network=host --name=mqtt --pod=hass -v /root/./mosquitto_data/mosquitto.conf:/mosquitto/config/mosquitto.conf:ro --mount type=bind,source=/etc/localtime,destination=/etc/localtime,ro  docker.io/eclipse-mosquitto:latest
podman create --network=host --name=zigbee2mqtt --pod=hass -e TZ=Europe/Amsterdam --mount type=bind,source=/root/./zigbee2mqtt_data,destination=/opt/zigbee2mqtt/data --mount type=bind,source=/run/udev,destination=/run/udev,ro --mount type=bind,source=/etc/localtime,destination=/etc/localtime,ro  --device /dev/ttyUSB0:/dev/ttyUSB0 koenkk/zigbee2mqtt
```

my configuration:
for mosquitto
```
listener 1883
socket_domain ipv4
allow_anonymous true
connection_messages true
log_timestamp true
```
for zigbee2mqtt (snipped away the devices and keys)
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost:1883'
  reject_unauthorized: false
serial:
  port: /dev/ttyUSB0
advanced:

  log_level: debug
```

when I run them I get this error:
```
Zigbee2MQTT:info  2021-08-08 15:51:58: Connecting to MQTT server at mqtt://localhost:1883
Zigbee2MQTT:debug 2021-08-08 15:51:58: Using MQTT anonymous login
Zigbee2MQTT:debug 2021-08-08 15:51:58: MQTT reject_unauthorized set false, ignoring certificate warnings.
Zigbee2MQTT:error 2021-08-08 15:52:08: Not connected to MQTT server!
```
the last line is then repeated multiple times

and mosquitto shows no connection attempts.

what I tried to debug:
I have created a new container with the mqtt cli from here https://github.com/hivemq/mqtt-cli
this container with the exact same networking parameters and I can successfully connect and send and receive messages. So the network connection to the mqtt broker works in my setup

I have build the image myself with buildah from the install instructions here https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html. this produces the exact same result

kernel is archlinux ```Linux home-assistant 5.10.52-6-ARCH #1 SMP Tue Aug 3 18:20:12 UTC 2021 armv7l GNU/Linux```

I am at a loss here. what ever I try I cant get it to work. Anybody got any help?

danieledwardgeorgehitchcock, 2021-08-10:
Try changing the server reference from localhost to the actual IP of your machine - does it work then?

fboerman, 2021-08-14:
hi @danieledwardgeorgehitchcock wow that indeed fixed it. the strange thing is this was not needed with other mqtt clients. weird, but thanks!

closed 2021-08-14 by fboerman