Koenkk/zigbee2mqtt #8310: 'tz.warning' must be wrong, cannot produce desired result

TheStigh, 2021-08-09:
Hi,

`tz.warning` (unknown author) should give you the possibilities to change Warning Mode, Strobe effect, Siren Level, Duration, Strobe Cycle and Strobe Level. I'm working on the Develco SIRZB-110. If I set the different values, it doesn't produce desired result/sometimes no result at all etc.

[URL to their tech doc](https://www.develcoproducts.com/media/1999/sirzb-110-technical-manual-smart-siren.pdf).

According to ZCL, I suspect this is caused by how we calculate the values to be sent.
As the ZCL says: 

> The Warning Mode and Strobe subfields are concatenated together to a single 8-bit bitmap field

Warning Mode are 4 bits, strobe on/off 2 bits and level are 2 bits.

_How I interpret it:_
Let's assume we want to use `burglar`, `no strobe` and `very_high` siren level.
This would be 1, 0 and 3 (from ZCL 8.4.2.3.1.2 - 8.4.2.3.1.4).
Binary this would be 0001, 00 and 11 (00010011). In DEC this is 19.

If I send `startwarninginfo: 19` I get a single beep per second, very low sound.

Changing it to just `high`, this should be 0001 00 and 10 (00010010)
DEC are 18.

sending `startwarninginfo: 18` doesn't change the volume, but a high/low siren (like an ambulance).

---

Current converter use Left shift (<<) and give different values.

I made a local converter and updated the `const info = const info = (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level] << 2);` where I changed how to calculate the level (I assume correctly).

Using same as above: `burglar`, `no strobe` and `very_high` siren level, would give value 28.
Now it sounds like a wake-up alarm beep-beep, beep-beep etc

Set level to `high` give value 24, but now there is no sound.

PS; also tested `tz.warning` on a Develco SMSZB-120 and the Climax SRAC-23B-ZBSR.
I see different devices use same config examples, but I'm really wondering if this has been tested at all? Ref the latest change you made on the strobe-part.

Koenkk, 2021-08-09:
I dont' have an alarm device to test this with. But maybe the order needs to be reversed. I've created the following example (save it as e.g. `test.js` and execute with `node test.js`):

```js
const assert = require('assert');

function calculateInfo(value, reverse) {
    const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
    const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
    const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};

    const values = {
        mode: value.mode || 'emergency',
        level: value.level || 'medium',
        strobe: value.hasOwnProperty('strobe') ? value.strobe : true,
        duration: value.hasOwnProperty('duration') ? value.duration : 10,
        strobeDutyCycle: value.hasOwnProperty('strobe_duty_cycle') ? value.strobe_duty_cycle * 10 : 0,
        strobeLevel: value.hasOwnProperty('strobe_level') ? strobeLevel[value.strobe_level] : 1,
    };

    if (reverse) {
        const info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
        return info;
    } else {
        // current behaviour
        const info =  (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level]);
        return info;
    }
}

// CURRENT
assert(calculateInfo({mode: 'stop', strobe: false, level: 'low'}) == 0b00000000);
assert(calculateInfo({mode: 'stop', strobe: false, level: 'medium'}) == 0b00000001);
assert(calculateInfo({mode: 'stop', strobe: false, level: 'very_high'}) == 0b00000011);
assert(calculateInfo({mode: 'stop', strobe: true, level: 'very_high'}) == 0b00000111);
assert(calculateInfo({mode: 'stop', strobe: true, level: 'medium'}) == 0b00000101);
assert(calculateInfo({mode: 'burglar', strobe: true, level: 'medium'}) == 0b00010101);
assert(calculateInfo({mode: 'emergency', strobe: true, level: 'medium'}) == 0b00110101);
assert(calculateInfo({mode: 'emergency_panic', strobe: true, level: 'medium'}) == 0b01100101);

// REVERSE
assert(calculateInfo({mode: 'burglar', strobe: false, level: 'low'}, true) == 0b00000001);
assert(calculateInfo({mode: 'burglar', strobe: true, level: 'low'}, true) == 0b00010001);
assert(calculateInfo({mode: 'burglar', strobe: true, level: 'very_high'}, true) == 0b11010001);
```

Can you try with `const info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);` ?

TheStigh, 2021-08-09:
I'm running the addon, from where should I run the test? Just Terminal or jump into the Z2M container ?

Koenkk, 2021-08-10:
@neugenn could you also try this change: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files ?

@TheStigh previously the order was MODE_STROBE_LEVEL, now it is LEVEL_STROBE_MODE

neugenn, 2021-08-11:
@Koenkk 
This change made my WD go haywire :(. Not working as expected. Why the 'reversed' computation for startwarninginfo ?

TheStigh, 2021-08-11:
> 
> 
> @Koenkk
> This change made my WD go haywire :(. Not working as expected. Why the 'reversed' computation for startwarninginfo ?

How did you trigger it ?

neugenn, 2021-08-11:
@TheStigh 
From the GUI.
Devices -> <WD_HERE> ->Exposes-> select mode, level, strobe etc. -> Apply.
Selecting a low sound level made the WD sound like high/very_high. Definitely not 'low'. (on low, my WD issues no sound alert)
Also, could not set WD strobe to on.

TheStigh, 2021-08-11:
@neugenn 

Works like a charm here, both with/without strobe.
This screenshot are from the Climax siren

[image]

neugenn, 2021-08-11:
[toZigbee.txt](https://github.com/Koenkk/zigbee2mqtt/files/6968335/toZigbee.txt)

@TheStigh 
Please find attached my local converter.
It contains some work in progress for Squawk command, but other that that, it is the converter available in version 1.21.0-dev (commit #a380df4f) (file extension changed from .js to .txt to be able to upload it)

TheStigh, 2021-08-11:
> 
> 
>
> What type of coordinator are you using ?

Slaesh

TheStigh, 2021-08-12:
> 
> 
> @neugenn can you confirm that with the current code (z2m 1.21.0 the log matches the values in the wireshark sniff). If yes, it means that the device of @TheStigh are actually not compliant to the zigbee spec and the tweak (https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files) should only be applied to these devices.

I strongly disagree! Develco (develcoproducts.com) are a Danish company, also making OEM. I would be surprised if they are not compliant, but iHorn are. Please also note that the Climax siren works as expected the this new code, not with the code in 1.21.0.

@neugenn Do you have documentation on the iHorn device?

neugenn, 2021-08-13:
@TheStigh I have no technical documentation. Just a user manual.
@Koenkk @TheStigh  Please find attached Z2M logs and its corresponding wireshark trace file. These are obtained using Z2M version 1.21.0-dev (commit #a3f22833). You can see that the log matches the sniff.
ZIgbee network key : 01:03:05:07:09:0B:0D:0F:00:02:04:06:08:0A:0C:0D 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6980622/log.txt)

TheStigh, 2021-08-13:
Well, I'm not going to argue with you guys. I do believe that the current change made by @Koenkk are the correct version;

**_I have tested this on:_**

**Develco SMSZB-120 Smoke Alarm**
Support `startWarning`, `warning duration`, `siren level`
All OK with new code
`siren level` do not work with 1.21.0 code

**Develco SIRZB-110 Smart Siren**
Support `startWarning`, `warning mode`, `siren level`, `warning duration`
All OK with new code
`startWarning` - not ok with 1.21.0, did only do a single or double beep
`warning mode` - not working with 1.21.0
`siren level` - not working with 1.21.0

**Climax SRAC-23B-ZBSR Smart Siren**
Support `startWarning`, `strobe`, `siren level`, `warning duration`
All OK with new code
`startWarning` - not ok with 1.21.0, did only do a single or double beep
`strobe`, `siren level`, `warning duration` - not working 1.21.0 (yes, even warning duration)

I've also for SIRZB-110 made the local converter for squawk using same reverse, works exactly as described in documentation.

All devices are certified Zigbee devices listed on zigbeealliance.org
- https://zigbeealliance.org/zigbee_products/smart-siren/
- https://zigbeealliance.org/zigbee_products/smoke-alarm-2/
- https://zigbeealliance.org/zigbee_products/ac-powered-zigbee-router-rm-23bzbsr/

I do not have the hw needed to re-program my cc2531 at my present location, so I cannot do any sniffing.

Koenkk, 2021-08-13:
@kirovilya @nurikk do you maybe have a warning device to test this with? TLDR; does the strobe/level and mode work correctly now or is a reversion like https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files needed?

neugenn, 2021-08-17:
> With the 1.21.0 code (master), can you please confirm
> 
> 1. Does warning mode work (choose different modes) ?
> 2. Does strobe turn on/off
> 3. Can you change volume of each warning mode?
> 4. Can you change duration?

@TheStigh please find below the answers:

1. The device outputs the same audio for all the warning modes (fire, burglar etc).
2. Strobe can be turned on/off successfully
3. Volume for warning mode is reflected in the device's sound output.
4. The warning duration is successfully set/changed.

TheStigh, 2021-08-17:
@Koenkk Should we make a `tz.warning_reverse` ?
I have two different siren's (Develco  Climax) working with the revised code.
@neugenn works with "old" code, not with revised
Two others have tested with the Tuya 0216 and no difference what ever code thy used.

Koenkk, 2021-08-17:
I've checked with wireshark and with the current implementation the values match as @neugenn described. Therefore I assume that the `SIRZB-110` interpets the spec incorrectly (but I'm not 100% sure).

@TheStigh https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files should fix it for the `SIRZB-110` please test

TheStigh, 2021-08-17:
> @TheStigh https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files should fix it for the `SIRZB-110` please test

@Koenkk Test OK.
Please change  
`if (['SIRZB-110'].includes(meta.mapped.model))  `
to 
`if (['SIRZB-110', 'SRAC-23B-ZBSR'].includes(meta.mapped.model))`

This has been tested to work. Still, I think the non-certified device from a china company are not following the standards, rather the Zigbee certified devices from Develco & Climax.

Can you merge https://github.com/Koenkk/zigbee-herdsman-converters/pull/2892/files with the above changes to dev, please ?

closed 2021-08-18 by Koenkk

Rockman18, 2021-09-02:
@Koenkk : Bitron Smabit AV2010/29A seems to suffer the same mistake. 
This could explain why we need this controlling chapter on the device https://www.zigbee2mqtt.io/devices/AV2010_29A.html#controlling

How can I check on a hassio normal (not edge) installation ? Override the existing converter ?