Koenkk/zigbee2mqtt #8313: Bulb Trust ZLED-RGBG6  interviewed as AduroSmart 81809/81813 before isn't working now and interviewing as a ZLED-TUNE9 on 1.21

SolovevVyacheslav, 2021-08-09:
## What happened

Bulbs "Trust ZLED-RGBG6"(not supported oficially) stopped working correctly. I can't switch on/off, change brightness , color.
Before 1.21 update they were recognized as AduroSmart 81809/81813(supported device) and worked perfectly (on/off, brightness, rgb color control)
Now they are recognizing as a Trust ZLED-TUNE9 (no rgd controll) and do not work in terms of control. But they can be repared and interviwed as Trust ZLED-TUNE9.

Z2M pairing logs here: https://pastebin.com/HaP4wKtA

Is there any way to apply converters from AduroSmart 81809/81813 manually or add them as a new device?
I can provide more information if needed. I just do not know what can be useful for now. Thanks!
## What did you expect to happen
I would like them to work like before =)

## How to reproduce it (minimal and precise)
Pair Trust ZLED-RGBG6 on 1.21 or update to 1.21 with bulb connected.

## Debug info
Zigbee2MQTT version: 1.21
Adapter hardware: 	CC2652P (Egony v4 usb stick)
Adapter firmware version:  20210319

SolovevVyacheslav, 2021-08-09:
Some additions:
Before repairing bulbs could be controlled through mqtt messages (even UI model name was wrong) but not z2m UI
After  repeiring, neither mqqt nor the user interface work
Direct bindings are still working. Atleast somehow Im able to switch on/off light in my WC room =)

Koenkk, 2021-08-11:
Looks the same as #8027, issue should be fixed in the latest dev branch now.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). If not, please post your data/database.db entry of your device in #8027

closed 2021-08-11 by Koenkk