Koenkk/zigbee2mqtt #8318: Tuya Smart Door Sensor (TY-ZDR06)

steemsjo, 2021-08-10:
## Tuya Smart Door Sensor
**Link:** https://www.aliexpress.com/item/1005002531475594.html
**Model:** TY-ZDR06
**Wireless:** Zigbee 3.0
**Picture:**
[image]

## Zigbee2Mqtt
**Zigbee2MQTT version:** 1.19.1 commit: 9bd4693
**Coordinator type:** zStack3x0
**Coordinator revision:** 20210120
**Frontend version:** 0.4.16

## Logs
```
Zigbee2MQTT:info  Device '0x00124b000706e837' joined,
Zigbee2MQTT:info   Starting interview of '0x00124b000706e837'
Zigbee2MQTT:warn  Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined',
Zigbee2MQTT:warn  Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
As a developer, I tried to follow the route of adding support for this, but I have no extra information like being described in the examples.

## data/database.db
``` json
{
  "id": 33,
  "type": "EndDevice",
  "ieeeAddr": "0x00124b000706e837",
  "nwkAddr": 59686,
  "manufId": 0,
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        3,
        1,
        1280
      ],
      "outClusterList": [
        0,
        3,
        1,
        1280
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "interviewCompleted": false,
  "meta": {},
  "lastSeen": 1628581933046
}
```

steemsjo, 2021-08-10:
Debug log when pairing:
```
Zigbee2MQTT:info  2021-08-10 14:01:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"MAG_1_FRONT_DOOR","ieee_address":"0x00124b000feda55a","status":"started"},"type":"device_interview"}',
Zigbee2MQTT:info  2021-08-10 14:01:00: Device 'MAG_1_FRONT_DOOR' joined,
Zigbee2MQTT:info  2021-08-10 14:01:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"MAG_1_FRONT_DOOR","ieee_address":"0x00124b000feda55a"},"type":"device_joined"}',
Zigbee2MQTT:info  2021-08-10 14:01:00: Starting interview of 'MAG_1_FRONT_DOOR',
Zigbee2MQTT:debug 2021-08-10 14:01:10: Received Zigbee message from 'MAG_1_FRONT_DOOR', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0,
Zigbee2MQTT:debug 2021-08-10 14:01:08: Received Zigbee message from 'MAG_1_FRONT_DOOR', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TYZB01_xph99wvr"}' from endpoint 1 with groupID 0,
Zigbee2MQTT:debug 2021-08-10 14:01:06: Received Zigbee message from 'MAG_1_FRONT_DOOR', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0203"}' from endpoint 1 with groupID 0,
Zigbee2MQTT:info  2021-08-10 14:01:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"MAG_1_FRONT_DOOR"},"type":"device_announced"}',
Zigbee2MQTT:info  2021-08-10 14:01:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"MAG_1_FRONT_DOOR","ieee_address":"0x00124b000feda55a"},"type":"device_announce"}',
Zigbee2MQTT:debug 2021-08-10 14:01:01: Device 'MAG_1_FRONT_DOOR' announced itself,
Zigbee2MQTT:debug 2021-08-10 14:01:00: Received Zigbee message from 'MAG_1_FRONT_DOOR', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0,
Zigbee2MQTT:debug 2021-08-10 14:01:00: Received Zigbee message from 'MAG_1_FRONT_DOOR', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0,
Zigbee2MQTT:info  2021-08-10 14:01:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"MAG_1_FRONT_DOOR"},"type":"pairing"}',
Zigbee2MQTT:info  2021-08-10 14:01:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"MAG_1_FRONT_DOOR"},"type":"device_connected"}'
```
I see the `"modelId":"TS0203"` coming by and saw in the issues that something was fixed for this in version 1.22 so I will wait for that release and see what it gives.

Koenkk, 2021-08-10:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7553

closed 2021-08-10 by Koenkk