Koenkk/zigbee2mqtt #8319: Help adding new thermostat ELKO ESH Plus Super TR RF PH

owangen, 2021-08-10:
I have this ELKO ESH Plus Super TR RF PH (model nr 4523430). It's a flooring heating Thermostat sold in Norway. 
It looks like it's supported in Deconz according to https://zigbee.blakadder.com/Elko_4523430.html (I have not tested myself), and I want to add support for it in zigbee2mqtt.

I'm not sure where to start for implementing support for it. I have earlier some experience adding different light support to zigbee2mqtt herdsman, but this thermostat is a little bit more complicated. So I need some help to where to start.

owangen, 2021-08-10:
I see that there has been some work on this issue before. https://github.com/Koenkk/zigbee2mqtt/issues/4856

docnannus, 2021-08-15:
Following. I have 5 of these. Currently just dumb thermostats due to no joy in deconz. Bought a stick from slaesh hoping for better support with mqtt.

owangen, 2021-08-16:
Great. I have some progress with supporting this thermostat thanks to the work done in #4856. So far I'm trying to get the data from the thermostat working (fromZigbee). Localtemp is working, but I need to create a custom converter to parse the custom attributes the ELKO uses. All this is documented well thanks to earlier work from #4856. Next step is to implement the toZigbee support, while i'm not sure what is possible to control... Need to dig into that

docnannus, 2021-08-16:
Would you share your code (and where I should put it)? New to mqtt as of yesterday, so that would save me some time. Might be worth looking into the work done  for support in deconz through Rest-API in terms of controllable attributes?

owangen, 2021-08-16:
So far I have used this https://gist.github.com/owangen/8fd0a49ec587fc793b5167a58f0b171a to enable it in Zigbee2mqtt. But I have not created a custom converter so there are many attributes not showing. 

### database.db entry for the Elko Thermostat
`{
   "id": 37,
   "type": "Router",
   "ieeeAddr": "0x000d6f000d95a73b",
   "nwkAddr": 37936,
   "manufId": 4098,
   "manufName": "ELKO",
   "powerSource": "Unknown",
   "modelId": "Super TR",
   "epList": [
      1
   ],
   "endpoints": {
      "1": {
         "profId": 260,
         "epId": 1,
         "devId": 769,
         "inClusterList": [
            0,
            3,
            513
         ],
         "outClusterList": [],
         "clusters": {
            "genBasic": {
               "attributes": {
                  "modelId": "Super TR",
                  "manufacturerName": "ELKO",
                  "powerSource": 0,
                  "zclVersion": 1
               }
            },
            "hvacThermostat": {
               "attributes": {
                  "1025": 1000,
                  "1026": "Super",
                  "1027": 1,
                  "1028": 15,
                  "1029": 0,
                  "1030": 1,
                  "1032": 0,
                  "1033": 2330,
                  "1041": 0,
                  "1042": 0,
                  "1043": 0,
                  "1044": 28,
                  "1045": 0,
                  "1047": 0,
                  "1048": 10,
                  "1049": 0,
                  "localTemp": 2670,
                  "occupiedHeatingSetpoint": 2200,
                  "systemMode": 1
               }
            },
            "hvacUserInterfaceCfg": {
               "attributes": {}
            }
         },
         "binds": [
            {
               "cluster": 513,
               "type": "endpoint",
               "deviceIeeeAddress": "0x00124b002390d981",
               "endpointID": 1
            },
            {
               "cluster": 0,
               "type": "endpoint",
               "deviceIeeeAddress": "0x00124b002390d981",
               "endpointID": 1
            },
            {
               "cluster": 3,
               "type": "endpoint",
               "deviceIeeeAddress": "0x00124b002390d981",
               "endpointID": 1
            }
         ],
         "configuredReportings": [
            {
               "cluster": 513,
               "attrId": 0,
               "minRepIntval": 0,
               "maxRepIntval": 3600,
               "repChange": 10
            },
            {
               "cluster": 513,
               "attrId": 18,
               "minRepIntval": 0,
               "maxRepIntval": 3600,
               "repChange": 10
            },
            {
               "cluster": 513,
               "attrId": 28,
               "minRepIntval": 10,
               "maxRepIntval": 3600,
               "repChange": null
            }
         ],
         "meta": {}
      }
   },
   "zclVersion": 1,
   "interviewCompleted": true,
   "meta": {
      "configured": 1159609792
   },
   "lastSeen": 1629149251848
}`

owangen, 2021-08-17:
I'm not quite sure how to proceed to converting the custom attributes that is reporting from the hvacThermostat cluster. I have looked at (what I think is) similar devices when it comes to converting attributes. For example Danfoss eTRV0100 Ally Thermostat. But can't get it working correctly. I don't have much time investigating into this, so any help on how to convert the customer attributes is appreciated.

owangen, 2021-08-18:
Heavily inspired from the Danfoss eTRV0100 Ally Thermostat and trying to get one of the custom attributes working (external temp).

What I'm I doing wrong here? 
What is exactly the configureReporting doing?

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    elko_thermostat: {
        cluster: 'hvacThermostat',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            //const result = converters.thermostat.convert(model, msg, publish, options, meta);

		if (msg.data.hasOwnProperty('elkoExternalTemp')) {
                    result[postfixWithEndpointName('external_temp', msg, model)] =
                       msg.data['elkoExternalTemp'];
            }
			
      
        },
    },
}

const definition = {
    zigbeeModel: ['Super TR'],
    model: '4523430',
    vendor: 'ELKO',
    description: 'ESH Plus Super TR RF PH',
    fromZigbee: [fzLocal.elko_thermostat, fz.thermostat, fz.thermostat_weekly_schedule, fz.hvac_user_interface],
    toZigbee:  [tz.thermostat_local_temperature],
    exposes: [e.local_temperature()],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
		const options = {manufacturerCode: 0x1277};
		
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat', 'genBasic', 'genIdentify']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatSystemMode(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatSystemMode(endpoint);	
        
        await endpoint.configureReporting('hvacThermostat', [{
                attribute: 'elkoExternalTemp',
                minimumReportInterval: 60, //constants.repInterval.MINUTE,
                maximumReportInterval: 62000, //constants.repInterval.MAX,
                reportableChange: 1,
            }], options);
        
	}
}
module.exports = definition;`

Error
`Zigbee2MQTT:error 2021-08-18 07:00:23: Failed to configure 'gang_termostat', attempt 1 (Error: ConfigureReporting 0x000d6f000d95a73b/1 hvacThermostat([{"attribute":"elkoExternalTemp","minimumReportInterval":60,"maximumReportInterval":62000,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4727,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:627:22)
    at Object.configure (/app/dist/util/externally-loaded.js:45:9)
    at Configure.configure (/app/lib/extension/configure.js:137:13)
    at Configure.onZigbeeStarted (/app/lib/extension/configure.js:105:17)
    at Controller.callExtensionMethod (/app/lib/controller.js:408:58))`

sluquet, 2021-08-19:
Would love to have these added. I have five of the Elko thermostats and have to run a separate deconz instance to get these integrated in Home Assistant.

owangen, 2021-08-19:
Good. We will get it working. 
I don't understand how I set up reporting.  If I want to read the attribute  1033: (with value 2330 from database.db) How to I configure it? It's the same attribute 0x409 (floor temperature sensor measurement) documented here https://github.com/prj84/com.elko/blob/Master/drivers/ESHSUPERTR/ST_Code/Elko%20Thermostat%20-%20Vendor%20Specific%20attributes.txt
and also same attribute here https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

Can someone give me a hint? Right now it's trail and error, and it takes to much time (since it's only error :-))

Koenkk, 2021-08-19:
> My understanding of this, after reading some code, is that you set up reporting (through configurereporting) for different attributes and then parse the result through fromZigbee converters. Is this correct @Koenkk ?

Exactly!

owangen, 2021-08-23:
Hello. Finally I have some progress. I'm now able to read the custom elko attributes. I have tried with external temp sensor and mean power attribute, and it looks to work fine. More mapping and testing to come. 
If someone is interested to test this, please see this external converter: https://gist.github.com/owangen/8fd0a49ec587fc793b5167a58f0b171a

owangen, 2021-08-24:
@docnannus Could you test the dimmer with the latest convert definition I provided? When it comes to other implementations I look at booth deconz implementation as well as the Homey app for ELKO devices

docnannus, 2021-08-24:
I've been meaning to try since your last post. Am on paternal leave so my days are unpredictable in regards to availability of time 😅

owangen, 2021-08-24:
@docnannus Perfect timing for tech tinkering. :-) Just give me hint when you have looked into it.

emilhetty, 2021-09-27:
Really nice to see that you have added support for this thermostat, great work!
I added all the Zigbee ID's earlier to the converter and was working on this, but got stuck as you did and had some other priorities. Didn't see this before now.
I have documentation, which I unfortunately are not allowed to share, for all the ELKO devices.
So if you plan to add other ELKO devices, don't hesitate to ask if I could help with anything.

owangen, 2021-09-27:
Nice @emilhetty, thanks for the prework, it helped me alot. I just have one Elko Zigbee device (this thermostat) unfortunately. You don't have by any chance some insight to the zigbee radio on this device, more precise firmware? My zigbee radio on the thermostat is not working anymore after a mistake trying to upgrade the zigbee firmware (reference to this post https://www.hjemmeautomasjon.no/forums/topic/1121-er-elko-zigbee-%C3%A5pen/?page=12&tab=comments#comment-91508). I'm suspecting different hardware versions and therefore maybe also different firmware versions for the zigbee radio.

emilhetty, 2021-09-27:
@owangen, Sorry, I only have the Zigbee ID's, datatypes and description.
It could be as you said about a different FW versions based on different HW.
Think the best is to just contact the elkosmart support, they have replied quickly before.

sveip, 2021-11-22:
Hi, I have a Elko Termostat that I have some problems with. 

I get this error when I try to add it: 

```js
Zigbee2MQTT:error 2021-11-21 21:16:56: Failed to configure '0x000d6f000d80f838', attempt 2 (Error: Bind 0x000d6f000d80f838/1 genIdentify from '0x00212effff0692d7/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:757:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:500:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/elko.js:65:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
```

Anyone know what to try to fix this?

owangen, 2021-11-22:
@sveip I'm not sure. What version does your device have? There are some version number shown at boot. Are your devices working @docnannus ?

docnannus, 2021-11-22:
6 thermostats working fine and dandy, v1.2.6R. Slaesh stick, deconz is somewhat experimental with zigbee2mqtt, no?

sveip, 2021-11-22:
I'm on v1.2.0 on this particular termostat. Hw rev is 1.3. (written on the
back sticker.)

owangen, 2021-11-22:
ok, then it could be something with the deconz zigbee2mqtt combination. I had also success with the Slaesh stick.