Koenkk/zigbee2mqtt #8321: Frient/Develco smoke detector - can't turn off alarm with switch

rindlerblabla, 2021-08-10:
## What happened
Since recently sirens are exposed to Home assistant as switches. Tried that function right now with some Frient smoke detectors. When I toggle the switch in Home assistant, the switch just toggles back directly to off. The alarm goes on, but I cannot turn it off with the switch, since it's already off according to the UI. Have some Tuya sirens as well, and they behave just fine.

`Debug Received MQTT message on 'zigbee2mqtt/Brandvarnare hall/set/alarm' with data 'START'`

## Debug info
Zigbee2MQTT version:  1.21.0-dev commit: 33ee9f7
Adapter hardware: Zzh 
Adapter firmware version: 20210120

rindlerblabla, 2021-08-19:
Updated to 1.21.0-dev commit: f9451ff which has several fixes to the smoke detector https://github.com/Koenkk/zigbee-herdsman-converters/pull/2895, however it didn't solve the HA switch.

@theStigh (sorry for tagging you on all my Develco/Frient issues), do you run Home assistant? Have you experienced the same?

TheStigh, 2021-08-19:
> 
> 
> Updated to 1.21.0-dev commit: [f9451ff](https://github.com/Koenkk/zigbee2mqtt/commit/f9451ffe22777c7aa2b0206625a689f3f2715387) which has several fixes to the smoke detector [Koenkk/zigbee-herdsman-converters#2895](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2895), however it didn't solve the HA switch.
> 
> @TheStigh (sorry for tagging you on all my Develco/Frient issues), do you run Home assistant? Have you experienced the same?

Yes, using Home Assistant.
I've made a custom switch sending the payload "alarm": "START" and "alarm": "OFF"

It appear as a switch in the Z2M UI -> exposes but does the same here.

@Koenkk  How can I resolve this? Remember earlier where a binary became a switch, can this be related to that change?

rindlerblabla, 2021-08-19:
Since I have several siren entities my plan is to put everyone in the same HA area. Then I think I would be able to use the service `homeassistant.turn_on` and trigger all sirens in the "siren area" without creating actions for all seperate sirens.

The switch in Z2M UI works just fine.

TheStigh, 2021-08-19:
Or you could use an automation to trigger all if a singe one are triggered (just work with actual smoke alarm, not the "alarm" :"START")

mamrai1, 2021-08-22:
Develco HESZB-120 fire detector doesn't expose siren as a switch in HA.. Am i missing something? Converters where updated only for the develco smoke detector?

rindlerblabla, 2021-09-27:
Still not solved. Am on 1.21.1-dev commit: 738b241.

@mamrai1 My heat detectors are exposes as switches in HA as well and have the same issue.

TheStigh, 2021-09-27:
> 
> 
> Since I have several siren entities my plan is to put everyone in the same HA area. Then I think I would be able to use the service `homeassistant.turn_on` and trigger all sirens in the "siren area" without creating actions for all separate sirens.
> 
> The switch in Z2M UI works just fine.

I think the `homeassistant.turn_on` only support zwave for now...

rindlerblabla, 2021-09-27:
> I think the `homeassistant.turn_on` only support zwave for now...

Obviously it is possible to run both `homeassistant.turn_on` and `homeassistant.turn_off` those switch entities already. It is working even though the switch show wrong status in HA UI.

artur3, 2021-10-01:
@rindlerblabla Can I ask how often your smoke sensors are reporting to zigbee2mqtt? 
I have one smoke and two heat sensors and log are full of messages from them. In two weeks reported voltage dropped from 3200 to 3000.

One sensor (grep enrolled):
`

info  2021-10-01 15:53:05: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 15:54:45: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 15:59:45: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:03:06: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:04:45: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:09:46: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:13:06: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:14:46: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:15:00: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:19:46: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:23:07: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:24:46: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:29:47: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:33:07: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',
info  2021-10-01 16:34:47: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled',

`
All of them:
`

info  2021-10-01 16:19:46: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:20:28: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:20:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:20:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:22:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:22:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:23:07: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:23:25: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:24:46: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:25:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:25:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:25:46: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:27:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:27:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:29:47: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:30:28: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:30:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:30:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:32:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:32:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:33:07: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:34:34: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:34:47: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:35:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:35:45: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:35:46: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:37:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:37:36: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:39:47: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
info  2021-10-01 16:40:28: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:40:46: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:40:46: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm2/enrolled', payload 'true'
info  2021-10-01 16:42:37: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:42:37: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/enrolled', payload 'true'
info  2021-10-01 16:43:08: MQTT publish: topic 'zigbee2mqtt/FrientSmokeAlarm1/enrolled', payload 'true'
`

rindlerblabla, 2021-10-01:
@artur3 Guess mine are updating with about the same interval. Set up a counter and in one hour it updated 21 times.

[image]

artur3, 2021-10-01:
@rindlerblabla  No problems with battery life?

rindlerblabla, 2021-10-01:
Dot know yet. Have only had them for about 3 months. The one reporting lowest value is now at 2900V.

mamrai1, 2021-10-01:
Maybe the temperature of the area you installed them changes too
frequently.. Check that. You can change minimum interval and/or temperature
change

rindlerblabla, 2021-10-01:
Seems like changing reporting interval from the UI is broken for me though. Just get error `Error Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Device 'Brandvarnare garaget/38' does not exist'`. Seems like it is adding the endpoint value but don't know if it really should.

artur3, 2021-10-01:
@mamrai1 Many reports have same temperature. 

zigbee-herdsman-converters/devices/develco.js 
`await reporting.temperature(endpoint2, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 10});`
How to understand min max values? Reports in intervals one to ten minutes?

mamrai1, 2021-10-01:
[image]

artur3, 2021-10-01:
Can this be changed directly in zigbee2mqtt configuration file? 
I'm using openHAB with zigbee2mqtt on Debian VM.

mamrai1, 2021-10-01:
Zigbee2MQTT has a built-in webbased frontend. To enable the frontend add the following to your configuration.yaml:

frontend:
  # Optional, default 8080
  port: 8080
  # Optional, default 0.0.0.0
  host: 0.0.0.0
  # Optional, enables authentication, disabled by default
  auth_token: your-secret-token

artur3, 2021-10-01:
Thanks. 
I must be blind. I looked at the zigbee2mqtt.io homepage so many times and didn't notice it.

artur3, 2021-10-02:
min       max         change
3600    21600       2     - this was set yesterday
3600    21600       5     - morning tests
7200    21600       0
7200    62000       0

Result, reports every 5min or less, regardless of settings:
`
info  2021-10-02 08:16:29: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:18:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:18:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:23:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:23:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:24:07: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:28:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:28:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:33:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:33:18: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:38:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:38:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:43:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:43:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:48:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:48:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:53:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:53:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:58:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 08:58:19: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.5'
info  2021-10-02 09:02:35: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:02:35: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:03:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:03:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:08:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:08:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:13:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:13:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:17:07: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:17:08: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:18:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:18:20: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:23:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:23:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:24:10: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:28:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:28:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:33:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:33:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:38:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:38:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:43:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:43:21: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:48:22: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:48:22: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:53:22: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:53:22: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.62'
info  2021-10-02 09:53:31: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
info  2021-10-02 09:53:39: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
[... cut ...]
info  2021-10-02 10:08:23: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
info  2021-10-02 10:13:23: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
info  2021-10-02 10:13:23: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
info  2021-10-02 10:13:47: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
info  2021-10-02 10:13:48: MQTT publish: topic 'zigbee2mqtt/FrientHeatAlarm1/temperature', payload '24.56'
`
Should I submit a bug report?

rindlerblabla, 2021-10-02:
It is also sending keep alive messages every 15 minute https://www.develcoproducts.com/media/1742/smszb-120-technical-manual-smoke-alarm.pdf

> 6.2 Normal – Keep alive
The device is sending a “keep alive” message to the PAN coordinator every 15 minute to verify that the device is still connected to the network.

Maybe @TheStigh knows more about whether it's a bug or not.

TheStigh, 2021-10-03:
This discussion past days obviously has nothing to do with the name of the thread. If you want me to check something, please make a question. 

Regarding Frient - I assume it is exactly same HW and uses same firmware as original Develco. What I don't know (as I don't have a Frient device, only original Develco's) are how it checks for firmware update and can read a Develco-firmware file.

When it comes to the Keep Alive, this is part of the IAS, and as @rindlerblabla referenced, it tells the PAN coordinator it is still there. 

When it comes to reporting, you can always customize these for your personal taste through the Z2M UI.

rindlerblabla, 2021-10-03:
@TheStigh No, sorry for not being clear, and it's off topic in this issue, but the question being raised is whether it is a bug that the device is updated more often than what the interval is set to even if no value has changed, see https://github.com/Koenkk/zigbee2mqtt/issues/8321#issuecomment-932711774.  It can also be the fact that the device send keep alive messages.

artur3, 2021-10-03:
@TheStigh I tried changing the reporting interval in the Z2M UI. Now it is less frequent, but far from configured. Please see my post above. 
How often do the original Develco sensors report? How long does the battery last? 

I will have another Frient smoke detector soon. I can send it to you.

@rindlerblabla Keep alive should be 15 minutes, mine reports every 5 minutes.

jamesonuk, 2021-11-02:
I worked around this by calling the MQTT_PUBLISH service.   So my button is
```
type: button
tap_action:
  action: call-service
  service: mqtt.publish
  service_data:
    topic: zigbee2mqtt/Smoke Alarm/set
    payload: '{"alarm": "START"}'
  target: {}
entity: switch.smoke_alarm_alarm
```
Change the payload to OFF to turn off the alarm.
Obviously you need to change the topic to match the name of your device

rindlerblabla, 2021-11-02:
> Obviously it is possible to run both `homeassistant.turn_on` and `homeassistant.turn_off` t on those switch entities already. It is working even though the switch show wrong status in HA UI.

And as stated in https://github.com/Koenkk/zigbee2mqtt/issues/8321#issuecomment-927782681 it is possible to to it like this in HA as well.

closed 2021-12-10 by github-actions[bot]

allgrinder, 2022-10-14:
I have the same Problem...