Koenkk/zigbee2mqtt #8322: Aqara Outlet H1 USB QBCZ15LM

xfzha, 2021-08-10:
## Information about the device + link
https://www.aqara.com/cn/productDetail/smart-wall-outlet-h1-usb

## data/database.db entry of the device
<code><pre>
{"id":37,"type":"Router","ieeeAddr":"0x04cf8cdf3c87bf4d","nwkAddr":34978,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug.sacn03","epList":[1,2,21,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.plug.sacn03","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":32,"stackVersion":2,"hwVersion":1,"dateCode":"06-22-2020"}},"genOnOff":{"attributes":{"245":117440520,"onOff":1}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[100,16,1,101,16,0,3,40,32,152,57,0,0,0,0,149,57,14,46,53,53,150,57,232,231,20,69,151,57,0,0,0,0,5,33,2,0,154,32,0,8,33,32,1,9,33,0,7,11,32,0]},"513":1,"515":1,"523":100,"550":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[6,0,3,4,5],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":42508800,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":2,"hwVersion":1,"dateCode":"06-22-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-31071145},"lastSeen":1628606053156}
</code></pre>

<pre>
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.plug.sacn03'],
    model: 'QBCZ15LM',
    vendor: 'Xiaomi',
    description: 'Aqara H1 outlet ZigBee',
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.ZNCZ15LM_overload_protection, fz.QBCZ15LM_button_switch_config, fz.xiaomi_switch_opple_basic],
    toZigbee: [tz.on_off, tz.xiaomi_switch_power_outage_memory, tz.xiaomi_led_disabled_night, tz.QBCZ15LM_button_switch_config, tz.ZNCZ15LM_overload_protection],
    meta: {multiEndpoint: true},
    endpoint: () => {
        return {'relay': 1, 'usb': 2};
    },
    exposes: [
        e.switch().withEndpoint('relay'), e.switch().withEndpoint('usb'),
        e.power().withAccess(ea.STATE), e.energy(), e.temperature().withAccess(ea.STATE), e.voltage().withAccess(ea.STATE),
        e.current(),e.power_outage_memory(), e.led_disabled_night(),
        exposes.enum('button_switch_config', exposes.access.ALL, ['relay', 'relay&usb']),
        exposes.numeric('overload_protection', exposes.access.ALL).withValueMin(100).withValueMax(2200).withUnit('W')
            .withDescription('power off automatically if the maximum power is exceeded')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
	},
};

module.exports = definition;
</pre>
All functions are working.
But the usb of this socket can be controlled separately, and its on/off is reported from endpoint2. I want to make a button controlled by usb alone. How can I do it?
Thanks.

Koenkk, 2021-08-10:
> I want to make a button controlled by usb alone. How can I do it?

I'm not sure what you mean, isn't that the `e.switch().withEndpoint('usb')`?

xfzha, 2021-08-10:
@Koenkk This doesn't work <pre>Error Entity '0x04cf8cdf3c87bf4d/usb' is unknown</pre>

Koenkk, 2021-08-11:
@xfzha should work after adding `usb` to this list: https://github.com/Koenkk/zigbee2mqtt/blob/70891eec39018303a43349c875ad04d44b3b43cf/lib/util/utils.js#L34

xfzha, 2021-08-11:
@Koenkk Yes, I have tried, add it after the <code>'relay'</code>, and then restart z2m, there are still errors.

Koenkk, 2021-08-12:
@xfzha what errors? can you provide the full debug log?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

xfzha, 2021-08-12:
I added <code>'usb'</code> to the back of the <code>'relay',</code>
<pre>
const endpointNames = [
    'left', 'right', 'center', 'bottom_left', 'bottom_right', 'default',
    'top_left', 'top_right', 'white', 'rgb', 'cct', 'system', 'top', 'bottom', 'center_left', 'center_right',
    'ep1', 'ep2', 'row_1', 'row_2', 'row_3', 'row_4', 'relay', 'usb',
    'l1', 'l2', 'l3', 'l4', 'l5', 'l6', 'l7', 'l8',
    'l9', 'l10', 'l11', 'l12', 'l13', 'l14', 'l15', 'l16',
    'button_1', 'button_2', 'button_3', 'button_4', 'button_5',
    'button_6', 'button_7', 'button_8', 'button_9', 'button_10',
    'button_11', 'button_12', 'button_13', 'button_14', 'button_15',
    'button_16', 'button_17', 'button_18', 'button_19', 'button_20',
    'button_light', 'button_fan_high', 'button_fan_med', 'button_fan_low',
    'heat', 'cool', 'water', 'meter', 'wifi',
];
</pre>
Then I click on/off
<pre>
debug 2021-08-12 23:07:09: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c87bf4d/usb/set' with data '{"state_usb":"ON"}'
info  2021-08-12 23:07:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c87bf4d/usb"},"type":"entity_not_found"}'
error 2021-08-12 23:07:09: Entity '0x04cf8cdf3c87bf4d/usb' is unknown
</pre>

Koenkk, 2021-08-12:
Can you try sending either:
- `zigbee2mqtt/0x04cf8cdf3c87bf4d/usb/set` `{"state":"ON"}`
- `zigbee2mqtt/0x04cf8cdf3c87bf4d/set` `{"state_usb":"ON"}`

xfzha, 2021-08-12:
<code>zigbee2mqtt/0x04cf8cdf3c87bf4d/usb/set {"state":"ON"}</code> the debug log:
<pre>
debug 2021-08-12 23:44:30: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c87bf4d/set' with data '{"state_usb":"ON"}'
error 2021-08-12 23:44:30: No converter available for 'state_usb' (ON)
</pre>

<code>zigbee2mqtt/0x04cf8cdf3c87bf4d/set {"state_usb":"ON"}</code> the debug log:
<pre>
debug 2021-08-12 23:45:05: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c87bf4d/usb/set' with data '{"state":"ON"}'
info  2021-08-12 23:45:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c87bf4d/usb"},"type":"entity_not_found"}'
error 2021-08-12 23:45:05: Entity '0x04cf8cdf3c87bf4d/usb' is unknown
</pre>

danieledwardgeorgehitchcock, 2021-08-13:
Could it be that you only have endpoint 1 in your configure section?

I am not entirely sure what your config entries are actually doing?

xfzha, 2021-08-13:
@danieledwardgeorgehitchcock No, if I change the <code>usb</code> to the <code>left right center</code>, it can be on/off normally

Koenkk, 2021-08-13:
@xfzha ive tried to reproduce this but failed to do so, the following test passes fine: https://github.com/Koenkk/zigbee2mqtt/commit/19b5ecf61771f579b7850ff7efacc987791c1495 after changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/5c0f0194d0ab27b22eebf4b31f9560da1e9f8588/devices/xiaomi.js#L430 to `return {'system': 1, 'usb': 2, 'right': 3};`

xfzha, 2021-08-14:
@Koenkk It’s strange, I swapped the relay and usb, The <code>relay</code> controls the usb port, and the <code>usb</code> controls the relay port. Now the usb can be normally on/off by the <code>relay</code>, but the relay cannot be controlled by the <code>usb</code>. It seems that my <code>'usb'</code> doesn’t work. I changed <code>zigbee2mqtt/lib/util/utils.js</code>, is this correct?

Koenkk, 2021-08-14:
Ah you should indeed modify `dist` (if you modify `lib` you need to run `npm run build` afterwards).

z-x-f, 2021-08-14:
Ok, I will create a pull request later, adding <code>'usb'</code> is also my create pr?

Koenkk, 2021-08-15:
no problem, assuming this can be closed now.

closed 2021-08-15 by Koenkk