Koenkk/zigbee2mqtt #8329: polling instead of assuming states

sygys, 2021-08-11:
I just started using Z2M for a few days now. (Been using ZHA for over a year). For my test I added some devices in all classes to see if Z2M is something i could use. But only after a few days i ran in so much thins i dont like about it that im allready considering to revert back.

One of the reasons i really dont like Z2M is the way it communicates with devices. ZHA polls devices to get the actual state where Z2M assumes states in which sometimes (especially when devices have a bad connection) the state presented in Home assistant isnt even close to what the actual state is. And that state stands indefinately if its not updated manually. That is not how a reliable smarthome should work. No device or service should assume a state in any way.

A good example happened yesterday when my hue motion sensor somehow fell of the grid after detecting motion. So this state was updated to Z2M and was the last received state when i fell off the grid. Then a few hours later it came back up. Z2M registered it back on the network and reported the last received state back to home assistant which was 'on' (motion detected) Because we were allready in bed the automation kicked in for detecting motion and the lights went on. But because the motionsensor never registered any motion in the first place it didnt need to update its state to 'off' in which Z2M kept the motion detected state. This resulted in my light being on all night. 

With ZHA it would have polled the last state after a disconnect and would have reported off to home assistant.

Besides this i allready had a couple of times that i changed the color of a light in which a light did not respond to it. This also happens to ZHA sometimes but with ZHA if the light isnt responding the color will bounce back to the one it really is. Z2M just keeps the new color in home assistant even when the command never made it to the light. So for example home assistant is reporting the light to be purple while it actually is green. The bad part is that it will continue to report it purple until you manually set it otherwise.

All of this are lights so not a real problem you would think. But i also have a watering system for my garden with a zigbee relais. If this turns on and after say 30 min i turn it off and Z2M only reports its off but actually the command didnt make it to the relay this would mean the watering system will stay on indefinately and there is no way for me to see that because it will report off in home assistant.

I really cant use a system that is so unreliable. Z2M really needs descent polling or else this whole system is useless to me.

sygys, 2021-08-11:
I just learned that battery powered devices never poll not even on ZHA. THen this would have been bad luck.

But still it would be nice if lights etc. were polled to get the right data. And also the relais switch.

danieledwardgeorgehitchcock, 2021-08-11:
What you are requesting is already a feature in Zigbee2MQTT please see [here](https://www.zigbee2mqtt.io/information/availability.html)

This enables Zigbee2MQTT to poll devices at a given frequency, and will also refresh its state. As for Zigbee2MQTT attempting to set a state, failing, and then retaining the failed state, this should not happen as Zigbee2MQTT awaits a state response from a device (fromZigbee) and will use that as it's retained value if a response is not received, it should revert back to the initial values before the set command.

closed 2021-09-19 by github-actions[bot]