Koenkk/zigbee2mqtt #8330: Smart plug TuYa Model Zigbee TS0121

amy92000, 2021-08-11:
Hello
I have some zigbee plugs of the TuYa TS0121 model
but I have no energy back.
when I look at the documentation I see this
thanks in advance.

sensor:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    value_template: "{{ value_json.energy }}"
    unit_of_measurement: "kWh"
    device_class: "energy
[image]
[image]

Plawasan, 2021-08-11:
FWIW I have the same power socket and I do get an energy readout... (still no way to reset it though..)

From HA Device info (don't have access to z2m dashboard right now):
10A UK or 16A EU smart plug (TS0121_plug)
by TuYa
Firmware: Zigbee2MQTT 1.21.0

Fridge Current 0.37 A
Fridge Energy 29.48 kWh
Fridge Power 89 W
Fridge power outage memory On

GSzabados, 2021-08-15:
Something is fishy here. I have just lost the energy reading for that plug. (Or I just noticed it...)

@Plawasan, can you restart HA and Z2M just for a check that it shows up normally after?

GSzabados, 2021-08-15:
@Plawasan - Never mind, I think, I've found the issue.

https://github.com/Koenkk/zigbee2mqtt/issues/8201

@amy92000 - Are you still on HA 2021.7.X as I am? I haven't updated to 2021.8.X yet, but that seems to fix my Unknown issue and I guess it should fix yours as well.

amy92000, 2021-08-16:
This is what HA my outlet is connected via the ZigbeeLinker plugin (zigbee2mqtt)

GSzabados, 2021-08-16:
> This is what HA my outlet is connected via the ZigbeeLinker plugin (zigbee2mqtt) Jeedom

Ah, OK. By the way, have you used that plug at least that much that it would show any energy?

Try to re-interview the plug as well.

stevetz, 2021-08-17:
I'm also having the same issue - re-interviewed. run for 30 hours. I also changed the poll interview to 30s, with no luck changing. energy is showing as `null`

amy92000, 2021-08-18:
it's the same with me too "energy": null

amy92000, 2021-08-18:
I found the solution you have to go to Z2M selection not taken and you go to the report you will have to add commands as in the photo and then you [work.]
[image]

amy92000, 2021-08-18:
you just have to add up to line in z2m reports and you would have the value Energy
[image]

stevetz, 2021-08-18:
@amy92000 merci, this works perfectly!

amy92000, 2021-08-18:
Super contente pour toi👍👍😂😂

csszokolai, 2021-08-21:
> I found the solution you have to go to Z2M selection not taken and you go to the report you will have to add commands as in the photo and then you [work.]
> [image]

It works for me! Thank you!

closed 2021-09-28 by github-actions[bot]