Koenkk/zigbee2mqtt #8331: TS0216 Siren doesn't complete interview

Morphy99, 2021-08-11:
## What happened
Tried pairing and removing several times and interview doesn't complete successfully however the device appears to function correctly
[image]

## What did you expect to happen
Interview successful. 

## How to reproduce it (minimal and precise)
Press Vol+ & Vol- for 5 seconds, device starts interview, press vol buttons to keep alive.

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: CC2530
Adapter firmware version: 20201128

##Log
```
info  2021-08-11 14:32:35: Device '0x847127fffe0fff2c' joined
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x847127fffe0fff2c","ieee_address":"0x847127fffe0fff2c"},"type":"device_joined"}'
info  2021-08-11 14:32:35: Starting interview of '0x847127fffe0fff2c'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x847127fffe0fff2c","ieee_address":"0x847127fffe0fff2c","status":"started"},"type":"device_interview"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x847127fffe0fff2c"},"type":"device_connected"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x847127fffe0fff2c"},"type":"pairing"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:35+01:00"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:35+01:00"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x847127fffe0fff2c","ieee_address":"0x847127fffe0fff2c"},"type":"device_announce"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x847127fffe0fff2c"},"type":"device_announced"}'
info  2021-08-11 14:32:35: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:35+01:00"}'
info  2021-08-11 14:32:40: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:40+01:00","linkquality":89}'
info  2021-08-11 14:32:40: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:40+01:00","linkquality":89}'
info  2021-08-11 14:32:40: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:40+01:00","linkquality":89}'
info  2021-08-11 14:32:40: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:40+01:00","linkquality":89}'
info  2021-08-11 14:32:41: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:41+01:00","linkquality":89}'
info  2021-08-11 14:32:41: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:41+01:00","linkquality":89}'
info  2021-08-11 14:32:41: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:41+01:00","linkquality":89}'
info  2021-08-11 14:32:41: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:41+01:00","linkquality":89}'
info  2021-08-11 14:32:41: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:41+01:00","linkquality":89}'
info  2021-08-11 14:32:42: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:42+01:00","linkquality":89}'
info  2021-08-11 14:32:42: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:42+01:00","linkquality":89}'
info  2021-08-11 14:32:42: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:42+01:00","linkquality":89}'
info  2021-08-11 14:32:42: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:42+01:00","linkquality":89}'
info  2021-08-11 14:32:42: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:42+01:00","linkquality":89}'
info  2021-08-11 14:32:43: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:43+01:00","linkquality":89}'
info  2021-08-11 14:32:43: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:43+01:00","linkquality":89}'
info  2021-08-11 14:32:43: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:43+01:00","linkquality":89}'
info  2021-08-11 14:32:43: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:43+01:00","linkquality":89}'
info  2021-08-11 14:32:44: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:44+01:00","linkquality":89}'
info  2021-08-11 14:32:44: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:44+01:00","linkquality":89}'
info  2021-08-11 14:32:44: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:44+01:00","linkquality":92}'
info  2021-08-11 14:32:44: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:44+01:00","linkquality":89}'
info  2021-08-11 14:32:44: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:44+01:00","linkquality":89}'
info  2021-08-11 14:32:45: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:45+01:00","linkquality":89}'
info  2021-08-11 14:32:45: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:45+01:00","linkquality":89}'
info  2021-08-11 14:32:45: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:45+01:00","linkquality":89}'
info  2021-08-11 14:32:45: MQTT publish: topic 'zb2mqtt/TRV Kidsroom', payload '{"battery":10,"battery_low":false,"current_heating_setpoint":20,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":false},"eurotronic_system_mode":1,"eurotronic_trv_mode":null,"eurotronic_valve_position":null,"last_seen":"2021-08-11T14:32:44+01:00","linkquality":31,"local_temperature":20.5,"occupied_heating_setpoint":20,"pi_heating_demand":0,"system_mode":"auto","unoccupied_heating_setpoint":16,"update":{"state":"idle"},"update_available":false}'
info  2021-08-11 14:32:45: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:45+01:00","linkquality":89}'
info  2021-08-11 14:32:45: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:45+01:00","linkquality":92}'
info  2021-08-11 14:32:46: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:46+01:00","linkquality":99}'
info  2021-08-11 14:32:46: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:46+01:00","linkquality":115}'
info  2021-08-11 14:32:46: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:46+01:00","linkquality":70}'
info  2021-08-11 14:32:46: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:46+01:00","linkquality":105}'
info  2021-08-11 14:32:47: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:47+01:00","linkquality":92}'
info  2021-08-11 14:32:47: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:47+01:00","linkquality":94}'
info  2021-08-11 14:32:47: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:47+01:00","linkquality":84}'
info  2021-08-11 14:32:47: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:47+01:00","linkquality":107}'
info  2021-08-11 14:32:47: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:47+01:00","linkquality":110}'
info  2021-08-11 14:32:48: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:48+01:00","linkquality":102}'
info  2021-08-11 14:32:48: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"last_seen":"2021-08-11T14:32:48+01:00","linkquality":102}'
[... cut ...]
info  2021-08-11 14:34:16: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"alarm":true,"battery":null,"last_seen":"2021-08-11T14:34:15+01:00","linkquality":81,"volume":0}'
info  2021-08-11 14:34:23: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"alarm":true,"battery":null,"last_seen":"2021-08-11T14:34:23+01:00","linkquality":65,"volume":0}'
info  2021-08-11 14:34:24: MQTT publish: topic 'zb2mqtt/0x847127fffe0fff2c', payload '{"alarm":false,"battery":null,"last_seen":"2021-08-11T14:34:24+01:00","linkquality":65,"volume":0}'
```

Koenkk, 2021-08-12:
Can you provide the herdsman debug logging when pairing this device?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Morphy99, 2021-08-13:
Thanks @Koenkk Here you go 
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/6981812/z2m.log)
Think I got it all.

Koenkk, 2021-08-13:
I've added some more logging to debug this further, can you provide a new log on the latest dev branch?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Morphy99, 2021-08-15:
So I removed and re-joined it on the latest dev and it joined successfully. Unfortunately I didn't capture the log of this and the second time around it again joined without completing the interview successfully again. Also the volume buttons still didn't seem to do anything but I'm not sure if this is the device as I can't see any messages when I press them. Here's the log anyway.
[z2m-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/6987929/z2m-debug.log)

Morphy99, 2021-08-15:
Just tried it again and interviewed successfully again. I think it works when I remove it and then enable joining, it joins again without having to hold the volume buttons down. Here's that log.
[z2m-debug-success.log](https://github.com/Koenkk/zigbee2mqtt/files/6987941/z2m-debug-success.log)

Koenkk, 2021-08-15:
In your log I see the interview is still failing, but I found the issue now and fixed it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Morphy99, 2021-08-16:
I can confirm it's now interviewed successfully. I just need to figure out how to change volume from HA and why the volume buttons aren't working now. I will open new issues if needed for that though. [I can see](https://community.home-assistant.io/t/do-you-have-a-zigbee-siren-and-running-z2m-looking-for-ppl-to-do-a-test/329419/5) someone is already working on developing this device further.

closed 2021-08-16 by Morphy99