Koenkk/zigbee2mqtt #8333: Siterwell GS361A-H04 new whitelabel

eskholm, 2021-08-11:
I’ve bought a “Nedis ZBHTR10WT“ smart radiator themostat. It’s a Siterwell GS361A-H04 in disguise, but isn’t found by z2m as supported. It presents itself as:

Zigbee model 'TS0601' and manufacturer name '_TZE200_hhrtiq0x'

Adding the manufacturer name to 

zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/siterwell.js

makes the device supported and fully functionable.

robertalexa, 2021-08-11:
@eskholm I have created a PR on your behalf based on the above information you have provided
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2900

Hope this helps

Koenkk, 2021-08-12:
Added thanks to @robertalexa 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-12 by Koenkk