Koenkk/zigbee2mqtt #8334: TS0201 pairs OK but never reports.

Westcott1, 2021-08-11:
CC2530,  version: {"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}
Zigbee2MQTT version 1.21.0-dev

I've got 3 of these that announce themselves as TS0201
[https://www.aliexpress.com/item/1005002830299478.html?spm=a2g0s.9042311.0.0.4d324c4dj3Jwry](url)
They pair successfully, but never report any data.
Are they a different sort of TS0201, because they should also report illuminance.
My other normal round TS0201 works correctly.

mrTuomoK, 2021-08-12:
Please check this issue / thread --> [issue 7741](https://github.com/Koenkk/zigbee2mqtt/issues/7741) (I guess you have this device?)

Westcott1, 2021-08-12:
Thanks, that's it exactly!
What a long thread with no real solution.
I'll close this.

closed 2021-08-12 by Westcott1