Koenkk/zigbee2mqtt #8339: Switch 2 and 3 not working on multi relay ORVIBO RL804QZB

ffragnol, 2021-08-12:
**What happened:**
---

After paring my new Multi-functional 3 gang relay ORVIBO RL804QZB, i can use switch 1 properly, but i cant use switch 2 and switch 3.

[image]

**What did you expect to happen :** 
---

Using the switch 2 and 3 without errors.

**Debug info :** 
---

Zigbee2mqtt version :  1.21.0-1
 Orvibo hardware info : 
[image]
[image]

When i try to use switch 2 or switch 3  i get this errors : 

```
debug Received MQTT message on 'zigbee2mqtt/Portail_coulissant/l2/set' with data '{"state_l2":"ON"}'

error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:170:50) at Controller.callExtensionMethod (/app/lib/controller.js:408:81) at runMicrotasks (<anonymous>) at processTicksAndRejections (internal/process/task_queues.js:95:5))

debug Received MQTT message on 'zigbee2mqtt/Portail_coulissant/l3/set' with data '{"state_l3":"ON"}'

error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:170:50) at Controller.callExtensionMethod (/app/lib/controller.js:408:81) at runMicrotasks (<anonymous>) at processTicksAndRejections (internal/process/task_queues.js:95:5))
```

When i use switch 1 : 

```
debug Received MQTT message on 'zigbee2mqtt/Portail_coulissant/l1/set' with data '{"state_l1":"ON"}'

info MQTT publish: topic 'zigbee2mqtt/Portail_coulissant', payload '{"linkquality":191,"state_l1":"ON","state_l2":null,"state_l3":null}'

```

ffragnol, 2021-08-12:
When i check my database DB it seems epList contains not correct values  :

`{"id":30,"type":"Router","ieeeAddr":"0x00124b0022315e44","nwkAddr":17901,"manufId":0,"manufName":"ORVIBO","powerSource":"Mains (single phase)","modelId":"82c167c95ed746cdbd21d6817f72c593","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":515,"inClusterList":[0,258,5,4,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"82c167c95ed746cdbd21d6817f72c593","manufacturerName":"ORVIBO","powerSource":1,"zclVersion":1,"appVersion":"\u0001\u0001","hwVersion":1,"dateCode":"2018-08-16","swBuildId":"v3.1.06"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff060e3a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":"\u0001\u0001","hwVersion":1,"dateCode":"2018-08-16","swBuildId":"v3.1.06","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1628703775476}
`

fmplaw, 2021-08-29:
If anyone has any tips (or advice) to solve my problem, I'm interested! ;)

ffragnol, 2021-09-29:
I still have this problem.

ffragnol, 2021-11-15:
still have the problem !

closed 2021-12-23 by github-actions[bot]

djfluo, 2022-05-16:
Hello, I have exactly the same problem :(