Koenkk/zigbee2mqtt #8349: GL-SD-001 brightness times out

Julusian, 2021-08-12:
## What happened

Changing brightness of GL-SD-001 works, but command times out causing other changes to be queued.
Additionally, the updated brightness is not reported back to mqtt

## What did you expect to happen

Brightness to change succesfully

Querying the currentLevel in the dev console immediately reports the correct value (once the timeout has happened)

## How to reproduce it (minimal and precise)

In the frontend or equivalent, set the brightness
This will be immediately visible
Trying to change anything again will time out

## Debug info
Zigbee2MQTT version: 1.21.0 commit: 70891ee
Adapter hardware: CC2652R
Adapter firmware version: zStack3x0 20210120

Hopefully I didn't miss anything in this log, there was some other traffic in the middle that I have tried to filter out.
https://pastebin.com/XDy5VxPn

database.db entry:
```
{"id":20,"type":"Router","ieeeAddr":"0xcc86ecfffebfb2f5","nwkAddr":47694,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-SD-001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-SD-001","manufacturerName":"GLEDOPTO","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"20210322","swBuildId":"10451201"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":126}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021600203","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"20210322","swBuildId":"10451201","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1628784701651}
```

Julusian, 2021-09-13:
I believe still a problem

w00dwind, 2021-10-16:
Same issue. 
`Publish 'set' 'brightness' to 'Mom_main' failed: 'Error: Command 0x84fd27fffebb91b3/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43398 - 1 - 180 - 8 - 11 after 10000ms)'`

misternikitos, 2021-10-30:
I have the same problem.

Koenkk, 2021-10-30:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

misternikitos, 2021-10-31:
@Koenkk, now works perfect. Thanks!

wurststulle, 2021-10-31:
works perfect here, thanks!

closed 2021-10-31 by Koenkk

smily555, 2023-10-22:
Hi, I have the same problem with Gledopto GL-SD-003P

smily555, 2023-10-23:
This also fixed the problem for me. Can you put this line in the next branch.
` meta: {disableDefaultResponse: true},`

Koenkk, 2023-10-23:
@smily555 integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

smily555, 2023-10-23:
Thanks, all good!