Koenkk/zigbee2mqtt #8350: Aqara DJT11LM - no action value

cyablo, 2021-08-12:
Hi, I recently got a pair of Aqara DJT11LM vibration sensors to track my garage door and paired them to Z2M Version 1.21.0 .(home assistant addon). The x/y/z angles and strength attributes work nicely, but action doesn't.

It's eiter shown as "null" or "empty_string":

[image]

[image]

Zigbee debug messages:

> Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"85":1}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"85":1}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"1283":7,"85":2}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"85":3}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"1288":[64949,3604566]}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"1285":10878976}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"85":3}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"1288":[1361,3276861]}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Garagentor', type 'attributeReport', cluster 'closuresDoorLock', data '{"85":1}' from endpoint 1 with groupID 0

Because every other attribute works, I suspect the problem not to be my setup.

Regards

Daniel

danieledwardgeorgehitchcock, 2021-08-13:
The action field is momentary meaning that it reports the value when the action happens, then reverts back to the NULL or Empty String.

If you look at your debug logs, when you are tilting or shaking the device, you should see something come through

Koenkk, 2021-08-13:
@danieledwardgeorgehitchcock is right, this is expected behaviour when the Home assistant integration is enabled. Added this to the FAQ: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/FAQ.md#why-is-the-action-property-always-empty

closed 2021-08-13 by Koenkk