Koenkk/zigbee2mqtt #8351: Daily database corruption

a8ree, 2021-08-12:
## What happened

Virtually every day my devices vanish. When I look at my DB size it has shrunk from 29k to 3k. I have to stop the service and revert the database to a backup and then restart Zigbee2MQTT to restore service

## What did you expect to happen

I would expect that the database would be maintained and would not reset

## How to reproduce it (minimal and precise)

Run as normal - wait 24 hours and almost guaranteed to happen

## Debug info

https://pastebin.com/A22NYG3w

Zigbee2MQTT version: 1.20.0 commit: 41b67fdd (Windows)
Adapter hardware: 
Adapter firmware version:

Note: I am running NSSM on Windows to run as a service

Koenkk, 2021-08-13:
Could you share the herdsman debug log of the moment this happens?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

a8ree, 2021-08-15:
I think my issue is due to nssm. So I have moved to Linux and haven't had a reoccurrence. Thanks @Koenkk

closed 2021-08-15 by a8ree