Koenkk/zigbee2mqtt #8354: Expose Hue Power On state in Z2M frontend

Xaldimo, 2021-08-12:
## Is your feature request related to a problem? Please describe
It would be helpful if the Hue Power On settings could be set through the Z2M frontend

## Describe the solution you'd like
In the Expose section a sub-section in regards to the Power On settings, similar as the other settings.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Since the Hue has now the Power On settings, which can be set with a MQTT command, I was wondering if it is possible to expose it by default in the Z2M frontend and able to change the settings through it.

danieledwardgeorgehitchcock, 2021-08-13:
This is already available

For white ambience bulbs, you should have something like `color_temp_startup`

I am just looking in to what the equivalent exposed entity would be for colour bulbs and will update this comment once I have found it

Xaldimo, 2021-08-13:
I have just found that on the non color bulbs, but it doesn't have the option as described on the webpage where it states:

"hue_power_on_behavior": "on",          // default, on, off, recover

So I can't set it to 'recover' to go back to the last known state.

danieledwardgeorgehitchcock, 2021-08-13:
That should be labelled as `previous` or it is in mine anyway...

Xaldimo, 2021-08-13:
I must be getting blind. Have seen it now :smiley: and also seen it on the color bulbs. So will close this one. Sorry for this.

closed 2021-08-13 by Xaldimo

Final-Hawk, 2021-08-16:
Awesome, I didn't see that option before for my colour bulbs. Any chance [the white varient](https://www.zigbee2mqtt.io/devices/8718696449691.html) of the Philips Hue bulbs can be added to have this option as well?

GingerAdonis, 2022-08-29:
It would be great if this issue can be re-opened as one-color Hue lights do not expose the `hue_power_on_behavior` option to the frontend.

propi62, 2022-11-05:
yes, please make it work for non color bulps. i have 20 hue non color bulps