Koenkk/zigbee2mqtt #8355: Nue / 3A Dimmer Switch ( HGZB-04D ) - Transition command not working

NapsterAU, 2021-08-12:
The HGZB-04D is apparently supposed to support transition command but does not work.
Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/HGZB-04D___HGZB-4D-UK.md

Can confirm the command does work on the RGBW controller
Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/3A12S-15.md

NapsterAU, 2021-08-18:
Using brightness_step along with transition works but I still can't get the "state" or "brightness" topics to respect the transition option.

OzGav, 2021-08-22:
I have created a Nue device before but I won't say I am an expert! I have looked at the converter device definiton which is 
```
    {
        zigbeeModel: ['FB56+ZSC05HG1.0', 'FNB56-ZBW01LX1.2', 'LXN56-DS27LX1.3', 'LXN60-DS27LX1.3'],
        model: 'HGZB-04D / HGZB-4D-UK',
        vendor: 'Nue / 3A',
        description: 'Smart dimmer wall switch',
        extend: extend.light_onoff_brightness({disableEffect: true}),
        whiteLabel: [{vendor: 'Sunricher', model: 'SR-ZG9001K8-DIM'}],
    },
```
light_onoff_brightness code is here https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js
It has code relating to transitions within. So it should work.  I would suggest starting by changing the logging to debug and send some on and off commands to the switch with a transition and see what actually gets sent and see if you can norrow down where it is failing...

closed 2021-09-29 by github-actions[bot]