Koenkk/zigbee2mqtt #8358: Group Home Assistant Auto-discovery Update Home Assistant System ID

Danit2, 2021-08-13:
## Is your feature request related to a problem? Please describe
The function that groups are automatically displayed in the HA is great.
However, if a group is adjusted, this is not adopted in the HA.

## Describe the solution you'd like
A slider with "Update Home Assistant System ID" would be great, as is the case with the devices.

Koenkk, 2021-08-13:
> However, if a group is adjusted, this is not adopted in the HA.

What do you mean with "adjusted"?

Danit2, 2021-08-13:
Sorry i mean „edited“.

Koenkk, 2021-08-13:
Can you be more concrete here, what do you mean with edit? Add a device, change name?

Danit2, 2021-08-13:
Ahh, Sorry.

I mean, when i editing a name of a group it don't changes in HA.
Also when i delet a group it don't delet them in HA. 

Nice woud be the same as by the devices. with the slider :"Update Home Assistant System ID"

Sorry for my english.

Koenkk, 2021-08-15:
Agree, will add this later.

closed 2021-09-22 by github-actions[bot]