Koenkk/zigbee2mqtt #8359: FUT035Z Working Converter Addition

fluffymadness, 2021-08-13:
This is the CCT model of the led controller. This is what I needed to add for the device to work
```
,
{
        fingerprint: [{modelID: 'TS0502B', manufacturerName: '_TZ3210_frm6149r'}],
        model: 'FUT035Z',
        description: 'Dual White LED controller',
        vendor: 'Miboxer',
        extend: extend.light_onoff_brightness_colortemp(),
 }
```

fluffymadness, 2021-08-13:
right now i'm looking into how I can remove the color temp startup feature since this seems to do nothing

danieledwardgeorgehitchcock, 2021-08-13:
I think you are looking for the disableColorTempStartup option

fluffymadness, 2021-08-13:
@danieledwardgeorgehitchcock 
Thanks that did the job, everything is fully working now.

danieledwardgeorgehitchcock, 2021-08-13:
@fluffymadness do you want me to add this and create a pull request for you?

fluffymadness, 2021-08-13:
I've already made one, but thx for the offer.

closed 2021-08-13 by fluffymadness