Koenkk/zigbee2mqtt #8364: Renaming device breaks processing report messages to Home Assistant

tabnul, 2021-08-13:
**Issue:** In some cases, after renaming a device in Z2M, reporting messages do not get processed in HA. (last_seen and other properties not updated in HA, but they do get updated in Z2M). 

**Scenario:**
Today i joined 5 ikea bulbs. 2 of them i renamed (checking ' rename device in HA ' ). After this they stopped getting updates in Home assistant by reporting messages. Only if i switch the device manually via Z2M or HA , home assistant gets updated.

I've had comparable issues in the past and i thought i had a way of working to prevent this, restarting Home Assistant a minute after renaming the device. It dit not work today.
Also , another workaround was to remove the device, and rejoin it, change to a completely new name.

I am not sure whether this is an HA or Z2M issue. There is something going wrong when renaming the device. 
**To verify, i just remamed the two bulbs to their original name: reporting to HA is fixed! This proofs to me there is a structural issue here. **

Does anyone has a clue what happens here?

By the way , earlier i reported an issue which i think has the same root cause. I thought this was device specific but here i also renamed the device a few times even; https://github.com/Koenkk/zigbee2mqtt/issues/7646

zigbee2mqtt version : 1.19.1
homeassistant version: 2021.8.1

tabnul, 2021-08-13:
now i renamed the devices back to the not-working name, now they are reporting...

the behaviour seems very inconsistent.

Koenkk, 2021-08-28:
Just for trying, does this issue fix itself when removing that from your `configuration.yaml`?

tabnul, 2021-08-28:
> Just for trying, does this issue fix itself when removing that from your `configuration.yaml`?

Hi, yes that fixes the issue . Renamed it now 6 times without issues.
It makes sense i guess since there is no dependency on the availability topic anymore.

Koenkk, 2021-08-29:
Found the issue, I will fix it in https://github.com/Koenkk/zigbee2mqtt/issues/6281 (will take some time since I'm currently doing big refactoring there)

Koenkk, 2021-09-01:
Should be fixed now in the latest dev, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Koenkk, 2021-09-02:
Could you provide the debug logging of this?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.