Koenkk/zigbee2mqtt #8376: Cannot control devices after 24hrs of operation on ZZH CC2652R

gkyprios, 2021-08-15:
## What happened
I updated the firmware on my Electrolama ZZH CC2652R to 20210708 and HASS Addon to 1.21.0-1. I did two restarts are per the doco and everything appeared to be working at first. However after about 24hrs of operation I cannot control any devices and the logs fill with errors. I can get all devices working again if I pull the usb stick and restart the addon. If I just restart the addon it makes no difference. I have to pull the stick as well.

## What did you expect to happen
I expect the Addon to run without problems.

## How to reproduce it (minimal and precise)
Wait 24hrs then the logs start filling up with errors.

## Debug info
Zigbee2MQTT version: 1.21.0-1
Adapter hardware: CC2652R
Adapter firmware version: 20210708

Here are some log errors. The issue seems to start 22-23hrs later. I suspect an issue with the stick itself or more accurately, the new firmware. I emailed Electrolama 5 days ago and I'm waiting for a response. It was working flawlessly before I updated it. I've tried flashing the stick twice and both times it reports success. 

Zigbee2MQTT:error 2021-08-10 17:41:04: Publish 'set' 'state' to 'Ensuite George 1' failed: 'Error: Command 0x00178801035639c0/11 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

Zigbee2MQTT:error 2021-08-10 17:41:10: Publish 'set' 'state' to 'Ensuite George 2' failed: 'Error: Command 0x00178801035624e4/11 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

goliath888, 2021-08-15:
I am having the same problem (same error in logs) but with the cc2530. The problem has been issued here and it seems there are some people with the same problem.

#8282 

But I thoght that this problem was related to the old cc2530, but in your case, you are working with a more powerful CC2652R so not sure anymore. 

For now, my solution as mentioned in the other issue is to stop docker container, unplug USB, plug it again and start the container. The runs with no problem for a while...until that happens again.

xekil, 2021-08-15:
Hello, Following the update of the addon and the ZZH stick (20210120 -> 20210708) by scrupulously respecting the requested procedure, I have a similar problem.
Some devices stop responding for no reason.
In addition, the network map no longer displays the links (green line) between the coordinator and a large part of my peripherals while some work without problem

I had no problem before updating Z2M and ZZH :(

adamkrol93, 2021-08-17:
The same issue at my side. Yesterday I upgraded my stick and reconfigured the whole zigbee network. I even moved to some different channel (26) just to be sure that it doesn't overlaps the WiFi network (channel 2). After configuring everything again (~30 devices) everything worked like a charm. This morning nothing works. For every zigbee related action I am getting this in logs:
```
Zigbee2MQTT:error 2021-08-17 08:55:45: Publish 'get' 'state' to 'Kuchnia' failed: 'Error: Read 0x5c0272fffec9522c/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Zigbee2MQTT:info  2021-08-17 08:55:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'state' to 'Kuchnia' failed: 'Error: Read 0x5c0272fffec9522c/1 genOnOff([\"onOff\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Kuchnia"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2021-08-17 08:55:51: Publish 'set' 'state' to 'Kuchnia' failed: 'Error: Command 0x5c0272fffec9522c/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Zigbee2MQTT:info  2021-08-17 08:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Kuchnia' failed: 'Error: Command 0x5c0272fffec9522c/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Kuchnia"},"type":"zigbee_publish_error"}'
```

My stick is: cc2652p
Zigbee2Mqtt version: 1.18.1-1

YassineBenayad, 2021-09-17:
Hello
Probably you would be interested in issue #8282 we are closer to the solution there, I believe

Regards

gkyprios, 2021-09-17:
Thanks. I have been watching that ticket too.

closed 2021-10-25 by github-actions[bot]