Koenkk/zigbee2mqtt #8379: After updating from 1.19.0 to 1.21.0 i cannot start zigbee2mqtt

XerXes777, 2021-08-15:
## What happened
After updating from 1.19.0 to 1.21.0 i cannot start zigbee2mqtt.

## What did you expect to happen
log from zigbee2mqtt: [log](https://pastebin.com/Heu0v7P9)
log from npm: [log](https://pastebin.com/VXmtT6HK)

my update routine:
`sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt
cp -R data/* /opt/data-backup
sudo rm -r /opt/zigbee2mqtt
cd ..
sudo git clone  --depth 1 --branch 1.21.0 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm ci --production`

## How to reproduce it (minimal and precise)
everytime when i will to update

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: Slaesh's CC2652RB
Adapter firmware version: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'

Koenkk, 2021-08-15:
After updating, make sure to execute a `npm ci`.

XerXes777, 2021-08-23:
> After updating, make sure to execute a `npm ci`.

I use the npm ci command.
And now it works.
thank you very much.

closed 2021-08-23 by XerXes777